subsurface/subsurface #349: Cylinder use column; only 2 values selectable

janmulder, 2017-04-24:
On the equipment tab, the cylinder use column dis (at some point in time) a selection of 4 options. OC, Dilulent, Qxygen, not used were possible. Now only OC and not used are selectable. Also here, not sure when it got broken.

Ghalr, 2017-04-24:
Also in Gas used (Dive planner points) only allow to choose AIR

Is it intended ? How do I set a Nitrox or Trimix dive ?

janmulder, 2017-04-24:
Fill in the O2 and He values.

sfuchs79, 2017-04-24:
I can remember this from @atdotde but I thought that it was somehow done in a way that selecting diluent and oxygen afterwards also still worked...
https://github.com/Subsurface-divelog/subsurface/commit/c1dc0c9ce0c0b8482dffba0b7becdec44d6f6bec
But when looking at the code today I'm not so sure any more?

atdotde, 2017-04-25:
True enough, by changing the behaviour of that column from "select from a drop down" to change by clicking, it is now impossible to select O2 or dilluent. But I would claim, at least in the planner those two options do not really make sense: First, we don't track gas consumption in CCR mode since we decided that predicting diluent use is completely impossible at any meaningful precision while tracking O2 would essentially just be tracking time times an guessed metabolation rate.

This use becomes "diluent" as soon as there is a non-zero set-point specified for any leg of the dive. So maybe we should change the string from "OC gas" to "used" (meaning "OC or diluent").

What exactly was it that you intended to do @janmulder ? Am I missing something?

janmulder, 2017-04-25:
@atdotde: I fully agree with the current working of the planner wrt. CCR gas use. Indeed, diluent is difficult to plan/estimate up front. Obviously, it is primarily a function of depth changes during the dive, where you have to correct the counter lung volume (talking from the mCCR concept here, eCCR can be more advanced). So I am not convinced that your statement "completely impossible at any meaningful precision" is fully right. For example, compare it to the GUE style minimum gas calculation. That is also a bit fuzzy calculation (fuzzy in the sense that the chosen input parameters, like SAC factor and problem solving time are open for discussion). This all said, I am not promoting a diluent estimate calculation, but if someone is tempted to add this, I would be happy to test/evaluate that. And further agree that O2 use is trivial (and for me a rule of thumb of 30bar/hour for a 3 liter tank is perfectly adequate).

Renaming the cylinder use to: used/not used seems more in line with the current working, but I do not oversee all consequences of dropping the current 2 options for dil and oxy. So definitely no hurry to change things.

And what was I trying to do? Well, I am trying to find an issue with my OSTC diving it in fixed setpoint mode, and not seeing the setpoint at the start of the dive. So definitely, nothing related to the planner. And in that process I stumbled upon the use column, wondering what it is used for (in general, not only in the planner). Seeing 4 options in the code, but also seeing only 2 selectable. Just that fact led me to open this issue, which I now consider closing.

sfuchs79, 2017-04-26:
I understand that this may not even be needed at the moment but isn't it that we easily can reenable this?
Easy as in "even I could do this" ;-)
Or do I miss s.th.?
https://github.com/sfuchs79/subsurface/commit/091361642c0743e5a24d6baaf899aa594c078b97

sfuchs79, 2017-04-26:
BTW: One other thing: Diluent up to now was translated to "Verdünnung" in the German translation. For sure not incorrect but sounds funny. I believe at Germany every CCR divers calls it "Diluent".

closed 2017-04-29 by janmulder