subsurface/subsurface #353: [Question?]Subsurface-desktop: New updates are not detected using a development version - What is the correct behaviour ?

Ghalr, 2017-04-27:
Hello.

I have not used Subsurface-desktop installed into one of my computer since long time.
I wanted to update for the latest version and I am not sure how updates are supposed to work exactly.
What is the correct behaviour ?

installed Subsurface version: 4.6.3-284-gd82e200d3583

Opened "Subsurface"
No "New update available" notification
Opened "About Subsurface" to check current version
Opened "Check for updates" saying me 
[You are running a development version; latest version is 4.6.4.0]   OK button
Nothing happen.

What Subsurface is supposed to do ?

Do users need to go into Subsurface website / github to manually download the update ?

- Would it be possible to get a direct download link at least to Subsurface website / github ?
- Do you plan to improve this "update" feature ?

dirkhh, 2017-04-27:
If you are on a development version, Subsurface assumes you know how to get a new version installed.
If you are on a Windows or Mac release version, you'll get a link to the latest release to download.

(and the automatic check for updates, assuming you enabled it, only runs every two weeks, so even if you are on a release version, it might take a short while before you are told that a new version is out)

And no, we will not be able to actually do in-place upgrades. We simply don't have the manpower to build and support such a feature across all the OSs that we support.

closed 2017-04-27 by dirkhh

Ghalr, 2017-04-27:
About that "automatic check 2 weeks" do there are a specific reason for 2-weeks ?
Thanks :)

dirkhh, 2017-04-27:
Yes. Again, based on feedback both on the mailing list and from some of our users who responded to questions about this. It is perceived as often enough to be useful, but not as constant to be annoying.
Remember, most of our users use the software only a few times a year (some only once or twice a year) when they go on a dive trip. Yes, there are a fair number of people who dive every week and use Subsurface more frequently, but even for them, missing an update by a week or two isn't really an issue. And if they care, they can follow us on Facebook, G+, Twitter, or subscribe to our RSS feed :-)