subsurface/subsurface #354: [bug] Importing Datatrack/WLog corruped

olivstdev, 2017-04-27:
Sometimes, when importing a Datatrack/WLog logbook file, not all dives are imported and some dives appeared as corrupted in subsurface, while they appeared good in Datatrack / WLog.
The problem come from a desynchronisation when reading the Datatrack/WLog file.
Cause : In datatrack.c source file, function dt_dive_parser, the number of samples is badly calculated in samplenum variable. Since it is an integer variable, the result of the division is truncated to floor. It should be rounded to the next integer value for correct samples number calculation.
As a result, there can be a missing dive sample not read by Subsurface.
If unfortunately this unread sample contains the byte 0xA0, Subsurface can't synchronize to the next dive in the file, because 0xA0 sample byte is badly interpreted as a dive separation marker.

mturkia, 2017-04-29:
Any change you could fix this and create a pull request? Seems you understand the issue and have the possibility to test the fix as well.

olivstdev, 2017-05-02:
Made a pull request.

neolit123, 2017-09-23:
closing as PR was merged.

closed 2017-09-23 by neolit123