subsurface/subsurface #359: [Idea]

atdotde, 2017-04-28:
In addition to pictures we could show stills from videos taken during the dive. With the advent of cheap video cameras those will become more common.

Adding this need a bit research with the Qt video capabilities. This would be a nice medium size project.

dirkhh, 2017-04-29:
Go for it :-)

bstoeger, 2018-02-18:
I looked a little bit into this. Playing videos should not be too much of a problem with `QMediaPlayer`. For extracting thumbnails from videos I see two options and I don't like either:

1) _Use Qt_: There is a rather clumsy, but in principle not too hard to implement way. Play it with `QMediaPlayer` and stream it to a customized `QAbstractVideoSurface`, which stops after capturing a frame. Sounds good, but has a big caveat: At least on Linux, this has to be run in the main thread owing to an old Qt bug: https://bugreports.qt.io/browse/QTBUG-39567 . Thus, we'd have to move the movie fro the worker thread to the main thread. This would probably mean a refactoring of the thumbnailing.

2) _Use external library_: OpenCV comes to mind. With the whole annoyance of adding another dependency.

atdotde, 2018-02-19:
Have you managed to play a video in Subsurface?

bstoeger, 2018-02-20:
> Have you managed to play a video in Subsurface?

Yes, that's surprisingly trivial by using `QMediaPlayer` + `QVideoWidget`. A somewhat more complex way is subclassing `QAbstractVideoSurface` and projecting the frames by hand.

To automatically open the system video player instead of the image viewer, no change whatsoever is needed.

Side-note: There seems to be no reliable way of getting the supported video formats of the platform, so I fear we'll have to hard-code them.

bstoeger, 2018-03-04:
Sadly, whereas video-playback works flawleslly in `QVideoWidget`, it doesn't work with `QGraphicsVideoItem` for me, which would be needed to have it play in the profile-graph. It just flashes up for a fraction of a second and then vanishes. It plays the sound though. My next attempt would be subclassing `QAbstractVideoSurface` and projecting by hand onto the DivePictureItem. But that sounds unnecessarily complicated. :(

commit a7643327 (2018-07-28): Dive media: put duration on video thumbnails

On the profile, the run-length of the videos is visualized by a bar.
Add the same information to video-thumbnails in the dive-photo-tab.
Though in this case, render it as text on top of the thumbnails.

Fixes #359
closed 2018-07-28 by dirkhh