subsurface/subsurface #362: Dive import, duration sometimes show seconds (even 00 seconds)

chagam, 2017-04-29:
After import, some times, dive duration is in format minutes:seconds, sometimes not.

On this screenshot, you can see dives 630, 640, 653 respectively have a duration of 3m20s, 8m00s, 13m00s while all the others just have a number of minutes with no seconds. The time format should be constant. Either minutes or hours:minutes or hours:minutes:seconds

[image]

Ghalr, 2017-04-29:
I would prefer in minutes.

dirkhh, 2017-04-29:
This is tricky. Because different divers want different things. Freedivers want minutes and seconds. Cave divers want hours and minutes. Recreational divers most likely would be fine with minutes only.
I'd have to look at the code, but IIRC right now we give you hours when it's more than 60 minutes and we give you seconds when it's fewer than 15 minutes.
Yes, of course. We could add yet another setting. But remember, every setting makes the software harder to use for a beginner, requires more testing, more translation, introduces more bugs. So I'm not too keen on that.

sfuchs79, 2017-04-30:
For one part of this I see a simple solution w/o additional prefs option:
Instead of deciding based on a 15min threshold to use "MMmSSs" format for short dives we could do this based on dive mode for a existing dive. There we can select "Freedive" already today.

For minutes versus hours and minutes it might be very difficult to avoid a preferences option. Personally my friends and I meanwhile when we talk use to say: "Let's do a 75min/90min/100min/120min/... dive". So I would prefer this way. But I also know a guy who is regularly telling us that he did a 4 1/2 hours dive again... ;-)

sfuchs79, 2017-04-30:
Ah, we even have this already here but in "or" combination with the 15min threshold:
https://github.com/Subsurface-divelog/subsurface/blob/master/qt-models/divetripmodel.cpp#L356
Honestly I would remove the 15min thing.

Ghalr, 2017-04-30:
I wonder if it is not easy (reading) to use Hmins to calculate.
Also this is the same, to summarize how many hours you made each months, better to use "days" notation

sfuchs79, 2017-04-30:
I played around with different format options for scuba dives and freedives incl. taking into account the format of dive duration for the dive information tab and the question if I personally would also prefer to have the depth unit in the table.
My result is a little bit frustrating and I learn to understand @dirkhh better and better:
- I **am** able to find a solution and format I like :-)
- This even won't fit with what @Ghalr likes
- This for sure also won't fit with what thousands other users like
- No one really wants to have a config option for this minor thing

I'll wait for s.o. having a really great idea... unless s.th. asks me personally for my prefered option ;-)

chagam, 2017-04-30:
To me it's a question of consistency. It's not because I dive less than 15minutes or more than 1h that it should change the way duration is displayed. In settings, we already have units / Time units where we could put such setting. I understand that every setting makes the software harder to use. But in this case, we are just talking of a setting. Almost no algorithm/code as the actual code seems to be ready to such display change which means very small space for bug. I agree, it's more work for translators and for others. But isn't this software meant to evolve anyway ?

We all have different needs. Freedivers, divers, tec divers, etc. Actually, we tell them that we have one display that's not ideal but should be suitable for most and they have to deal with it.

To be 100% honnest, I can deal with it. Especially now that I know the 15min and 1h rule.

commit 92d24a29 (2017-05-07): Consistently show dive duration based on preferences

We now respect the settings in the preferences and also only show
the duration as minutes and seconds if the dive is a free dive.

Fixes #361
Fixes #362
closed 2017-05-07 by dirkhh