subsurface/subsurface #363: Dive import, hide or unselect short dives

chagam, 2017-04-29:
When importing lots of dives from the computer, you should be able to automatically hide some dives. 

As a Dive instructor, I still log my dives and when importing 50 dives, I have a lot of 1 or 2 min dive (shoot the lift bag of the anchor) or less than 3 meters (pool session). Of course, I can unselect them manually but that would be nice if I could automatically unselect the short and/or shallow dives. It could be a parameter in the settings

Ghalr, 2017-04-29:
Something like:

Import dive > depth      3m
Import dive > duration  5min

Totally agree

dirkhh, 2017-04-29:
While I usually reflexively push back against new features... yeah, I can see this one. But of course then we will immediately get an argument about the thresholds. Is 10ft/3m and 5 mins the right choice? Or do we need to make that configurable as well (which then turns this from one (or two?) checkbox(es) to two text fields)

atdotde, 2017-04-29:
Do you realise you can sort the dive list according to duration or depth by clicking on the respective table headers? Then it is trivial to select the shallowest or shortest dives and delete those. I guess this would be much easier then setting up a filter as you describe.

chagam, 2017-04-30:
@atdotde You are totally right. It was just an idea to make things easier for people who import dives often.

dirkhh, 2017-04-30:
As I said above - this one I can see. I don't "work" as a dive master much anymore, but for someone who teaches every week, this feature might make things a lot easier.