subsurface/subsurface #366: [Idea] merged dives with long surface interval

janmulder, 2017-04-29:
Again a bit special use case from multi sump cave divers (probably we have max 5 Subsurface users that this type of diving ...). Typically, the time above water is long enough to have the divecomputer splits the dive in more than one dive. No problem, as we can merge these. Sometimes that crossing of dry parts in a cave take much time, upto multiple hours, depending on the complexity and length of the dry cave in between.  Merging in these situations becomes a bit useless, as the profile graph contains a linear time axis, so an hour on the surface is a flat line of 60 minutes, compressing the interesting profiles . So it would be nice to split the time axis, or even leave out long surface intervals with some annotation on the profile graph showing a non linear time.

I know, a use case for a very limited amount of users, and probably not easy to implement.

dirkhh, 2017-04-29:
I agree with all parts of your analysis. Definitely something like that would be cool... it could be an option when you manually merge... "show surface interval / compress surface interval". 
BUT: that's a massive rewrite of some of our most complex code - the profile. And it touches a lot of different classes that today all make assumptions about time being linear...

So while I see this as a great idea, given the number of potential users and the fragility it will likely add to fairly complex code that is core to everything we do... Ummmm.

janmulder, 2017-04-29:
Yes, we are fully on the same page on this ... I would suggest to leave this issue open for some days or a week, so see if anybody, and specifically our Qt guru @tcanabrava, has some insight, that I am definitely missing on this.

janmulder, 2017-05-08:
Close for now. Too difficult to implement, for very limited amount of users.

closed 2017-05-08 by janmulder