subsurface/subsurface #367: Crash when doing "Import->GPS coord..." w/o userid in prefs (windows 10)

sfuchs79, 2017-04-30:
When I try to do "Import->GPS coord..." w/o having set a userid in prefs I get a crash under windows.

With release build I have a crash, with debug build as well because the assert in gpslocation.cpp#L44 is triggered:
https://github.com/Subsurface-divelog/subsurface/blob/master/desktop-widgets/subsurfacewebservices.cpp#L402
https://github.com/Subsurface-divelog/subsurface/blob/master/core/gpslocation.cpp#L44

[subsurface_GPS_getUserid.txt](https://github.com/Subsurface-divelog/subsurface/files/966610/subsurface_GPS_getUserid.txt)

dirkhh, 2017-05-06:
That should be easy enough to fix... do you want to submit a patch for it, or would you like me to fix it?

sfuchs79, 2017-05-06:
I also expect that thats easy - but unfortunately not for me ;-(
I had a look at the code but thats exactly the kind of C++ / OOP stuff where I'm immediately lost. Whats that "instance" stuff about? ;-)
Do I still have to say: Please you fix it!? I'll watch and learn...

dirkhh, 2017-05-06:
An instance variable is exactly what it sounds like. For some classes we want to have only one instance - we don't want to have multiple objects of this class, just exactly one. And to do that we use the static ::instance() member function (so this is a function that is connected to the class, not just to an object) which in return uses the static variable m_instance to hold a pointer to that one object.
I don't like that Q_ASSERT there... it would be better to simply instantiate the class when there is no instance. I'll try to fix this so you see what I mean.

commit 38af1f2a (2017-05-06): Check if we have an instance before dereferencing it

If we don't have a GPS service userid in the preferences and the GpsLocation class
isn't instantiated, this would cause a crash.

Fixes #367
closed 2017-05-06 by dirkhh