subsurface/subsurface #368: Add pictures to dive feature issues (Windows 10 and Linux)

sfuchs79, 2017-04-30:
Tested the really nice "picture" feature for the first time and stumbled across a couple of issues and bugs. I'm briefly going to list them here now also as a reminder for me. I can provide more details on request. I will also try to fix some of the items but with my limited experience... well you know...
- Mouse actions are pretty fragile mainly under Windows. E.g. the context menue doesn't open reliably on right-click into the picture "box". On Linux this works better. Also it is not 100% clear what actions should happen on different mouse actions. E.g. double-right-click does open a picture (as well as double-left-click). This doesn't feel correct. Selecting single/multiple pictures doesn't work.
- Right-click on a picture in profiles opens the picture (as well as left-click). This doesn't feel correct.
- In the context menu which opens under Linux and sometimes under Windows for right-click in the picture box, none of the four menu entries does work. This is also the case under Linux.
- For multiple of my JPG files evaluating the EXIF date/time seems to fail while Windows explorer is showing correct EXIF date/time.
- The list of files with (EXIF) date/time outside dive time should be displayed immediately and not after changing the time offset for the first time.
- The liste of files with (EXIF) date/time outside dive time needs a little bit of translation.
- The liste of files with (EXIF) date/time outside dive time needs a field/box with fixed height and automatically enabled vertical scroll bar. Otherwise a large liste of files will blow the UI window.

dirkhh, 2017-04-30:
A single "issue" with 7 different items is a bit frustrating as it doesn't really help us keep track of what's fixed and what's broken. And of course most of these need more information. Attach a picture where the EXIF date/time fails for us but works with Windows explorer. Show a screen shot of things that don't look right / lack translation.
I realize that you said this is in part to remind yourself, but simply from a process point of view it would be much easier to have "one problem per issue" and to get into the habit to allow someone else to immediately be able to reproduce the problem.

sfuchs79, 2017-04-30:
Yes, you are for sure right. I will cut this into pieces and add the details - just started. But please give me some time... I will have to do it step by step. As I said I even will try to fix a few items.
What would be great is a quick test for the UI/mouse event related stuff under windows by anyone and a quick heads up if s.o. sees similar things or if the issues only appear here.

dirkhh, 2017-04-30:
Thanks for splitting them up and posting sample pictures. That really helps. As has been obvious for the past few months, available time for me is very random... I don't think I'll have a lot of productive time until later next week

sfuchs79, 2017-05-01:
So here we now have the context menu and mouse event items left. See first comment in this conversation. I could split this also into two or more parts because we are talking here for sure about multiple items. But on the other hand if s.o. starts first with reproducing, these items could be checked in one block and for all of them we need s.o. with good Qt/UI skills to fix them.
With this I also want to say: I had a look at the code but I'm lost, sorry... 
@tcanabrava  Maybe Tomaz could have a look? :-)

sfuchs79, 2017-07-14:
@dirkhh recently fixed one item here but this is still left:

- Mouse actions are pretty fragile mainly under Windows. E.g. the context menue doesn't open reliably on right-click into the picture "box". On Linux this works better. Also it is not 100% clear what actions should happen on different mouse actions. E.g. double-right-click does open a picture (as well as double-left-click). This doesn't feel correct.

- Selecting single/multiple pictures doesn't work.

- Right-click on a picture in profiles opens the picture (as well as left-click). This doesn't feel correct.

neolit123, 2017-09-23:
hi @atdotde have you seen these comments from @sfuchs79 ?
i would comment myself, but i've never seen dive-pictures included in the profile or have looked at the code which handles that.

atdotde, 2017-09-23:
Cannot say anything about windows. Selecting multiple pictures would be great but opens a can of worms (keeping track of selected pictures). Maybe sometime in the future. Right click I can look at.

IzzySoft, 2017-10-23:
> Selecting multiple pictures would be great but opens a can of worms (keeping track of selected pictures).

Then, consequently, the "delete selected" context-menu item should be disabled until solved :wink:

> Right click I can look at.

Would be great – especially needed so one can at least delete single pictures.

PS: Most of the above seems to have been solved with v4.7.1 – except for the right click and image selection. I had exactly the same issues with v4.6.4 on Linux Mint 18.1.

closed 2017-12-01 by atdotde

sfuchs79, 2017-12-01:
@IzzySoft 
Finally found some time to fix the not working multi file select and context menu for the dive picture tab. Fix will be available in V4.7.5 which will be available soon and already in latest continuous build:
https://github.com/Subsurface-divelog/subsurface/releases

Maybe you could find some time to test it?

IzzySoft, 2017-12-01:
@sfuchs79 I'll try to check it ASAP – that is, as soon as 4.7.5 hits the repo and shows up at my end. Thanks a lot!

PR #887 (2017-12-01): Dive picture handling: Re enable multi select, improve mouse events

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Some improvements for the dive picture tab and dive pictures in profile:
- Bugfix mouse event in profile: Only Left-click will open picture
- Bugfix mouse events in picture tab:
  - Re-enable context menu (Windows bug mainly)
  - Re-enable multi select
  - Only double-left-click will open picture

Smaller  [... cut]