subsurface/subsurface #373: [Idea] Progress bar for any operation that takes some longer time

sfuchs79, 2017-05-01:
Create a standard progress bar pop up window or integrate a progress bar in the main window.

Show/activate this progress bar for any operation that takes a longer time. Up to now I mainly identified any image processing tasks as candidates but maybe there are other things.

mturkia, 2017-05-05:
Also import and export functions might take quite long. Unfortunately the XSLT transformation used for this does not really support reporting of progress... So version of the progress bar that just indicates something happening in this case would be in order.

dirkhh, 2017-05-06:
Yeah, having a busy indicator at least. I tend to have a reasonably fast machine, so I haven't seen anything that takes unreasonably long... can we collect specific actions that are good candidates for this? For example, exactly which image operations should have this? I think it would be reasonable to do this for all import and export operations that affect more than 5 dives or something. Or maybe just for all of them, but with a one second delay before it is shown. I need to re-read the code that I did for the cloud operations but I seem to remember there was something like this (but of course the other problem is one of not giving cycles to the main UI... I remember that that was a big issue with getting a decent progress report to work with the cloud operations...)

neolit123, 2017-10-09:
+1 for collecting a list of actions.

neolit123, 2017-10-09:
so the progress bar in the `MainWindow` can be potentially re-used for this.

for back-end work something like `QtConnectivity` is the correct choice, which creates a new thread from the thread pool and is flexible enough so that one does not need to subclass `QThread`.
but if one needs to update UI elements, the signals and slots mechanic needs to be used, so that the main thread can receive updates. i suspect that we need to update UI elements in some of the work that needs a progress bar so that complicates things.

@sfuchs79 
can you provide a quick list of operations that need a progress bar?

mikeller, 2026-10-18:
Hi @sfuchs79,

Following up on neolit123's question that never got answered: could you (or anyone else who wants this) put together a concrete list of the specific operations that should show a progress/busy indicator (large imports/exports, image/video thumbnail processing, cloud sync, etc.)? The implementation approach is already identified (QtConcurrent + signals/slots, reusing the existing MainWindow progress bar), the main gap blocking this has been exactly that list, which was never provided.