subsurface/subsurface #381: Some ideas to improve dive-planning functionality for technical diving

trimix25, 2017-05-04:
Hallo,

da ich leider keine andere Möglichkeit zur Kontaktaufnahme mit dem Entwicklerteam gefunden habe, probiere ich es über diesen Kanal.
Ich bin VDST/CMAS Trimix Instructor und habe mich umfassend mit den Funktionen des implementierten Tauchgangs-Planer beschäftigt. Ich bin begeistert! Saubere Arbeit. Euer Projekt kann locker mit den kommerziellen Tauchgangsplanern wie MultiDeco, V-Planner, GAP etc. mithalten.
Wie auch bei den kommerziellen Tauchgangsplanern fehlt jedoch noch die ein oder andere Funktion um das Tool in der Praxis sowie in Kursen Tauchverbands-übergreifend optimal einsetzen zu können.
Ich würde hierzu gerne einige Ideen einbringen, die sicherlich nur einen sehr geringen Programmieraufwand erfordern. Über eine Kontaktaufnahme würde ich mich sehr freuen:
E-Mail: jan[at]jan-network.de

Beste Grüße
Jan

mturkia, 2017-05-05:
Guten Abend,

Traditionally we have used developer mailing list subsurface@subsurface-divelog.org for discussing ideas and development in general. However, using issues on Github is also recommended so we can easily track the suggestions, ideas and bugs.

dirkhh, 2017-05-06:
@atdotde : hast Du Kontakt zu Jan aufgenommen?

atdotde, 2017-05-06:
I sent him an email with the offer to contact me directly. So far he did not. I guess for the time being, you can close this issue.

neolit123, 2017-09-23:
closing as suggested by @atdotde

closed 2017-09-23 by neolit123