subsurface/subsurface #387: Ratio iX3M connection fails after APOS 4.0.10 loaded

Exswede, 2017-05-07:
Hi,
After I have updated the iX3M from Ratio with the new APOS 4.0.10 s/w on the dive computer, I cannot download my logfiles from it.
Maybe it is because the USB driver files seems to be new as well, as I also had to download a new USB driver file and also Divelogger -4 program from Ratio

mturkia, 2017-05-07:
Can you enable the "Save libdivecomputer logfile" and "Save libdivecomputer dumpfile" options from the download dialog and provide the results? This should only produce these two debugging files, not actually import anything, but it will show us if the change is in the communication with the DC or if the data format has changed. (If the communication protocol has changed with the new driver, then Jef will hopefully instruct you how to sniff the USB traffic. BTW are you on Windows or Mac?)

Exswede, 2017-05-07:
I did enable both options, but only logfile is saved in the directory. I am pretty sure my iX3M is on COM1, but I captured t with COM4 too ( I cannot at the moment check, as the DiveLogger-4 Prg from Ratio crashes due to some Java issue )
I am running Subsurface on a Win-7 Ultimate 64-bit machine

[iX3M dload issue.zip](https://github.com/Subsurface-divelog/subsurface/files/981315/iX3M.dload.issue.zip)

 Weird, the system wouldn't let me attached the .txt file only so zip'd them up

mturkia, 2017-05-07:
Both of the logs are identical, and they don't show any communication occurring with the DC. I wonder if the port is still incorrect, or if the DC was not on download mode or something else. I'll send Jef (libdivecomputer author who writes the DC communication library) an email to make sure he spots this issue.

Exswede, 2017-05-07:
Thanks,
The iX3M is in PC Mode, showing that both cable and BT is ready, normally the Cable: shows DATA when something is talking to it.
I can see the FTDI driver that was installed by the Ratio update is a "USB Serial converter", this device is only listed int he device manager, when the iX3M is plugged into the USB port
[image]

mturkia, 2017-05-13:
Here is a response from Jef (unfortunately it was classified as spam so took me a while to notice):

The ix3m protocol doesn't support memory dumps. It will fail immediately with DC_STATUS_UNSUPPORTED with doing any communication. That's why the log file doesn't show anything useful. Instead of downloading a memory dump, just do a normal download with only the logfile checkbox enabled, and attach that logfile to the bug report. That should tell us more.

Exswede, 2017-05-14:
Attached is the results from just having the logfile selection selected, I did a run with either COM1 and COM4 selected, as both shows as options. Seems the COM4 is the "correct" one as it has more data, and when it was used the correct serial number of the unit, showed up in Subsurface.
[iX3M dload issue.zip](https://github.com/Subsurface-divelog/subsurface/files/999782/iX3M.dload.issue.zip)

jefdriesen, 2017-05-15:
Apparently the new firmware changed the data format. The size of the samples changed from 0x36 bytes to 0xC0 bytes, and that causes the download to fail.

I already patched a version of the dctool application available to fix the download part:

http://libdivecomputer.org/builds/experimental/windows/ix3m.exe

It's a command-line application which you need to run with these options:

ix3m -v -l ix3m.log -f idive -m 0x24 download -f raw -o dive.%n.bin COM3

Send back the ix3m.log and (if present) all dive.*.bin files.

I still need to figure out some way to detect the firmware version, to support both the old and new firmware. Can you tell me which version your ix3m has?

And I also need to verify whether the parser needs any changes or not. Most likely the first 0x36 bytes of the new sample is backwards compatible and the extra data contains new info, but I don't know for sure. To verify that, I'll probably need the database of the ratio software to compare against. The Ratio applications stores its data as a set of xml files in a directory named "dives-divelogger" in the installation directory. (This might be different depending on the windows version you have.) Just zip the entire directory.

Exswede, 2017-05-15:
Hi,
I ran the comand line tool, had to adjust it to COM4 at the end, below is the zip file of the output t created

My iX3m is running APOS: 4.0.10/011
[iX3M debiug output.zip](https://github.com/Subsurface-divelog/subsurface/files/1000483/iX3M.debiug.output.zip)

The DiveLogger program, is currently crashing, and Ratio working on a fix ( caused by some Java issue ) so cannot send more info right now...

fstengel, 2017-05-20:
Hello, I am seeing the same problem with my Mac (running OSX 10.10.5). So I downloaded the most recent mac version of the experimental software (http://libdivecomputer.org/builds/experimental/macosx/ix3m-v2) and ran it. The zipped archive contains the .bin files, the log as well as a copy of the dives-divelogger folder. The dives using APOS 4.0.10 are those numbered 16, 17 an 18.
 
File: [iX3M.zip](https://github.com/Subsurface-divelog/subsurface/files/1016248/iX3M.zip)

jefdriesen, 2017-05-21:
I received the necessary info about the new APOS4 firmware from DiveSystem/Ratio, and I'm working on a fix. The most important changes are already implemented in the ix3m-v2 build, but it's still work-in-progress because some details are not completely finished yet.

jefdriesen, 2017-05-24:
The ix3m fixes have been pushed to the libdivecomputer master branch.

Exswede, 2017-05-24:
Hi,
Will I get a notification to download a beta version to test, or just
through SubSurface normal update notification ?

janmulder, 2017-05-25:
@Exswede: first, the changes from Jef have to be merged into the Subsurface-branch of the (adapted) libdivecomputer. I have opened a pull request for that. But @dirkhh is very busy, so it might take some time before this is processed. After that, 2 ways are possible. Dirk can build a developer test version (and you will not get notified from Subsurface for that), or there will be an official 4.6.5 version of Subsurface (of which you will be notified).

Exswede, 2017-05-25:
OK, Thanks

cyberdiamond, 2017-06-13:
I have tested on the latest developer test version and can confirm it imports with the latest firmware installed.

This issue can now be closed.

janmulder, 2017-09-23:
Confirmed to work (but needs an official release, 4.7 for example).

closed 2017-09-23 by janmulder

Exswede, 2017-10-21:
Hi Guys,
Any ideas when this fix will be released in 4.7, or is there a Beta version I could download until 4.7 is fully released ?

Sorry for hassling you guys, just have a bunch of dives that it would be good to get off of my DC

Thanks in advance

dirkhh, 2017-10-21:
I hope to release 4.7 this weekend

Exswede, 2017-10-21:
Great !

Thanks Dirk