subsurface/subsurface #391: Cressi Leonardo doesn't import on Linux

torkve, 2017-05-07:
Still confirm the [report](http://trac.subsurface-divelog.org/ticket/707#no1)  from the old Trac: Cressi Leonardo still doesn't import dives on Linux.
Tested with 4.6.4-1~zesty.
Log:
```
INFO: Open: name=/dev/ttyUSB0
INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
INFO: Timeout: value=1000
INFO: RTS: value=1
INFO: DTR: value=1
INFO: Sleep: value=200
INFO: DTR: value=0
INFO: Sleep: value=100
INFO: Purge: direction=3
INFO: Write: size=8, data=7B3132334442417D
INFO: Read: size=0, data=
ERROR: Failed to receive the answer. [in cressi_leonardo.c:344 (cressi_leonardo_device_dump)]
```

neolit123, 2017-09-23:
@torkve sorry for the delayed reply (we are doing issue cleanup ATM). did you get this working?
@mturkia @glance- @jefdriesen any idea why this can happen?

glance-, 2017-09-24:
I have no clue. Maybe a timing issue on linux? Or maybe a usb-to-serial driver issue, for certain cables. 

@jefdriesen wrote all that backend, and I think he is the one who can figure this one out.

neolit123, 2017-09-24:
since this is an old issue, unless we get a response from @jefdriesen or @torkve within few days, i'd say we can close it - i.e. the user might have solved the issue.

torkve, 2017-09-24:
Hi guys, sorry for delay. Problem still remains, Ubuntu 17.04, kernel 4.10.0-35-generic #39-Ubuntu.
subsurface.log is the same as in initial report.
According to syslog, cradle was perfectly detected:
```
Sep 24 22:56:04 yaroot-46 kernel: [393903.187692] usb 2-2: new full-speed USB device number 13 using xhci_hcd
Sep 24 22:56:04 yaroot-46 kernel: [393903.334223] usb 2-2: New USB device found, idVendor=0403, idProduct=87d0
Sep 24 22:56:04 yaroot-46 kernel: [393903.334225] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 24 22:56:04 yaroot-46 kernel: [393903.334226] usb 2-2: Product: CRESSI PC INTERFACE 
Sep 24 22:56:04 yaroot-46 kernel: [393903.334227] usb 2-2: Manufacturer: CRESSI
Sep 24 22:56:04 yaroot-46 kernel: [393903.334227] usb 2-2: SerialNumber: 001612ET
Sep 24 22:56:04 yaroot-46 mtp-probe: checking bus 2, device 13: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2"
Sep 24 22:56:04 yaroot-46 mtp-probe: bus: 2, device: 13 was not an MTP device
Sep 24 22:56:05 yaroot-46 kernel: [393903.871040] usbcore: registered new interface driver usbserial 
Sep 24 22:56:05 yaroot-46 kernel: [393903.871053] usbcore: registered new interface driver usbserial_generic
Sep 24 22:56:05 yaroot-46 kernel: [393903.871064] usbserial: USB Serial support registered for generic
Sep 24 22:56:05 yaroot-46 kernel: [393903.874088] usbcore: registered new interface driver ftdi_sio
Sep 24 22:56:05 yaroot-46 kernel: [393903.874104] usbserial: USB Serial support registered for FTDI USB Serial Device
Sep 24 22:56:05 yaroot-46 kernel: [393903.874223] ftdi_sio 2-2:1.0: FTDI USB Serial Device converter detected
Sep 24 22:56:05 yaroot-46 kernel: [393903.874257] usb 2-2: Detected FT232RL
Sep 24 22:56:05 yaroot-46 kernel: [393903.874570] usb 2-2: FTDI USB Serial Device converter now attached to ttyUSB0
```

If I can provide any more debug, feel free to ask.

jefdriesen, 2017-09-24:
The dive computer doesn't respond to the commands that are send. Usually this indicates some external problem (e.g. low battery, bad or wet contacts, device not in PC mode, a problem with the driver, etc). But those are also the kind of problems that are very difficult to diagnose remotely. To be honest, I have no clue why it doesn't work.

We only know it works on Windows. So the dive computer (no low battery) and PC interface (no bad contacts) are probably fine, and obviously @torkve knows the correct procedure to initiate a transfer too (PC mode, although I don't even know if that is required for the leonardo). So basically the only thing that is left is the ftdi driver. But since ftdi is not some rare usb-serial chipset, I would be surprised if the problem is located there. And then I'm simply running out of ideas...

torkve, 2017-09-24:
Unfortunately I'm not familiar with driver debugging neither in windows, nor in linux. But if you can point me out where to start, I can attempt to investigate the problem myself.

salvadorcunat, 2017-09-24:
Blind shot, but ... has @torkve tried ftdi download instead of /dev/ttyUSB0 ? Just in case ...

torkve, 2017-09-24:
@salvadrcunat, how can I try it? I was able to read eeprom, but don't know how to download data via serial:
```
0 ➜ cat ftdi.cfg 
vendor_id=0x0403
product_id=0x87d0
self_powered=true
max_power=0
filename=data.bin
0 ➜ sudo ftdi_eeprom --read-eeprom ftdi.cfg 
FTDI eeprom generator v0.17
(c) Intra2net AG and the libftdi developers <opensource@intra2net.com>
FTDI read eeprom: 0
EEPROM size: 128
FTDI close: 0
0 ➜ hexdump -C data.bin 
00000000  00 40 03 04 d0 87 00 06  a0 2d 0c 00 00 02 98 0e  |.@.......-......|
00000010  a6 28 ce 12 43 10 05 00  0e 03 43 00 52 00 45 00  |.(..C.....C.R.E.|
00000020  53 00 53 00 49 00 28 03  43 00 52 00 45 00 53 00  |S.S.I.(.C.R.E.S.|
00000030  53 00 49 00 20 00 50 00  43 00 20 00 49 00 4e 00  |S.I. .P.C. .I.N.|
00000040  54 00 45 00 52 00 46 00  41 00 43 00 45 00 12 03  |T.E.R.F.A.C.E...|
00000050  30 00 30 00 31 00 36 00  31 00 32 00 45 00 54 00  |0.0.1.6.1.2.E.T.|
00000060  02 03 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 7c 40  |..............|@|
00000080
```

salvadorcunat, 2017-09-24:
hmmm. Not possible on 4.6.4.  you should try on a development version as ftdi code is recent.  Even then,  it's thought for direct usb connections, don't know if it will work with an interface.  @glance is the usual suspect for this feature ... I think it's worth to make a build and try.

torkve, 2017-09-24:
I've built latest subsurface with -DFTDISUPPORT=1, but I don't know what to do next? I tried just import from computer again and it's not working. How can I test ftdi download now?

salvadorcunat, 2017-09-24:
In the divecomputer download dialog, in the device, type ftdi instead of the usual /dev/ttyUSB0 (type it, it's not shown in the drop-down list).  Then click download as usual.

torkve, 2017-09-25:
I was able to start the download, running subsurface via `gksudo` (I assume, serial device isn't available for non-root by default).
There were some attempts, when it hanged on 96% each time, but then I killed the subsurface and started it again and download finished successfully.
[Here](http://paste.ubuntu.com/25613387/) is subsurface.log in case if you need it.

glance-, 2017-09-25:
You can give yourself access to the right usb character device, and then you don't need to be running it as root. You can find a udev rules file in your libftdi1 source tree, 99-libftdi.rules.

Given that, the actual connection works. Now the hard part is to figure out why you can't use the in-kernel-driver. Are you hit by some other odd program grabbing /dev/ttyUSB0 and sending crap to the dc, like modemmanager or something alike?

jefdriesen, 2017-09-25:
Checking whether another program is trying to grab /dev/ttyUSB0 is a very good suggestion! I completely forgot about that, but modemmanager is indeed know to cause problems due its automatic probing.

torkve, 2017-09-25:
Well, with `lsof` I see no process accessing ttyUSB0.
Also I see an appropriate udev-rule for modemmanager. /lib/udev/rules.d/77-mm-usb-serial-adapters-greylist.rules:
```
# FTDI-based serial adapters
#   FTDI does USB to serial converter ICs; and it's very likely that they'll
#   never do modems themselves, so it should be safe to add a rule only based
#   on the vendor Id.
ATTRS{idVendor}=="0403", ENV{ID_MM_DEVICE_MANUAL_SCAN_ONLY}="1"
```
I believe that it forbids modemmanager to probe it.
To be sure, I stopped modemmanager at all, but it doesn't help.

torkve, 2017-09-25:
According to sysdig, the only process accessing /dev/ttyUSB0 is subsurface.

torkve, 2017-09-25:
As of now, I found that connection via ttyUSB0 always times out right after sending initial header:
```INFO: Write: size=8, data=7B3132334442417D
INFO: Read: size=0, data=
ERROR: Failed to receive the answer. [in ../../source/src/cressi_leonardo.c:347 (cressi_leonardo_device_dump)]```
Increasing timeout to 5s doesn't help. The difference with FTDI is in the "Custom Write/Custom Read" used there, so I suppose that may be ttyUSB0-driver corrupts data somehow...

torkve, 2017-09-25:
I've finally made [two captures](https://github.com/Subsurface-divelog/subsurface/files/1330952/pcaps.zip) with wireshark: via ttyusb serial driver and via ftdi.
For ttyusb the header request is sent in packet `#68` (and no header response received), and for ftdi the header request sent in packet `#40` and response received in packet `#43`.

I don't know what should I do next, may be anyone can help me? Maybe some timings are wrong or some extra/missing data flags? Dunno what to look at.

torkve, 2017-09-30:
I've finally managed to find the problem source: it appears that Cressi Leonardo is unstable before first requests, and requesting the dump can cause it to produce garbage or produce just nothing.
Official Cressi software uses request of small data part as a ping, until it replies with correct data part. After that the device is stable and produces the proper output when the dump is requested.

So I made a patch for libdivecomputer, hopefully it will be accepted soon. With that patch connection via ttyUSB0 appears to be even more stable than via libftdi.

neolit123, 2017-11-09:
closing due to inactivity.
hopefully fixed in libDC.

closed 2017-11-09 by neolit123

paddatrapper, 2019-10-13:
I am still getting this error on Linux using the 4.9.3 AppImage. I get the same error on Windows if I do not let the official Cressi software connect to the cradle beforehand.

I can't find mention of the patch to libdivecomputer anywhere, so can't test if that works. Also using `ftdi` just presents the error "Unable to optn ftdi Cressi (Leonardo)"

```
Starting download from  ftdi
downloading only new dives
[0.000059] ERROR: No such file or directory (2) [in ../../src/serial_posix.c:295 (dc_serial_open)]
Finishing download thread: "Unable to open ftdi Cressi (Leonardo)"
```

glance-, 2019-10-14:
The official appimages ain't built with libftdi. so downloading via the ftdi custom serial won't work.

It looks like the thread never came to a conclusion:
http://libdivecomputer.org/pipermail/devel/2017-October/thread.html#1034

paddatrapper, 2019-10-14:
Ah OK. Yeah I'll follow up with that thread when I get a chance