subsurface/subsurface #398: MacOS Sierra

Paru78, 2017-05-13:
Subsurface never starts.
Don't know, what to do.

Application Specific Information:
terminating with uncaught exception of type NSException
abort() called
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSArrayM insertObject:atIndex:]: object cannot be nil'

dirkhh, 2017-05-13:
> On May 13, 2017, at 6:31 AM, Paru78 <notifications@github.com> wrote:
> 
> Subsurface never starts.
> Don't know, what to do.
> 
> Application Specific Information:
> terminating with uncaught exception of type NSException
> abort() called
> *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSArrayM insertObject:atIndex:]: object cannot be nil'
> 
> 

Which version are you trying to run? How are you trying to run it?

This works for quite a few people, so the obvious question is "what is different on your system"

Paru78, 2017-05-13:
This is my OS: macOS 10.12.4 (16E195)
And I am trying to run the 4.6.4 version of Subsurface. I downloaded it from your website and installed it on my Mac. Did nothing special. Because I never was able to get it running, I do not know, what I could change.

dirkhh, 2017-05-13:
> This is my OS: macOS 10.12.4 (16E195)
> And I am trying to run the 4.6.4 version of Subsurface. I downloaded it from your website and installed it on my Mac. Did nothing special. Because I never was able to get it running, I do not know, what I could change.
> 

I am running the exact same version of macOS on one of my Macs.
I went to https://subsurface-divelog.org/download and clicked on the Subsurface for Mac button which downloaded Subsurface-4.6.4.dmg
Double clicked on the downloaded file which opened a new Finder window
Double clicked on the Subsurface icon in that window
Pop up appears that say "Subsurface.app" is an application downloaded from the internet... - I click 'Open'
Subsurface runs fine.

I assume you followed the same steps? When you get the crash, could you capture the full output in the report window and send it to us? It's pretty lengthy, but it may contain information that helps us figure out what's different.

Thanks

/D

Paru78, 2017-05-13:
Yes. I did it exactly the same way. I will send you the crash report later.

Paru78, 2017-05-13:
Process:               Subsurface [668]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               4.6.4 (4.6.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [668]
User ID:               501

Date/Time:             2017-05-13 22:08:32.890 +0200
OS Version:            Mac OS X 10.12.4 (16E195)
Report Version:        12
Anonymous UUID:        72A84709-89EA-EE61-4908-E46A0C479E6B

Time Awake Since Boot: 310 seconds

System Integrity Protection: enabled

Crashed Thread:        10  Qt HTTP thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
terminating with uncaught exception of type NSException
abort() called
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSArrayM insertObject:atIndex:]: object cannot be nil'

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fffc319362b __exceptionPreprocess + 171
1   libobjc.A.dylib                     0x00007fffd83d01da objc_exception_throw + 48
2   CoreFoundation                      0x00007fffc30af50d -[__NSArrayM insertObject:atIndex:] + 1389
3   QtNetwork                           0x00000001054671b8 _ZNK15QSslCertificate6toTextEv + 55624
4   QtNetwork                           0x000000010546349c _ZNK15QSslCertificate6toTextEv + 39980
5   QtNetwork                           0x0000000105463687 _ZNK15QSslCertificate6toTextEv + 40471
6   QtCore                              0x0000000105e47d59 _ZN11QMetaObject8activateEP7QObjectiiPPv + 3113
7   QtNetwork                           0x000000010543285e _ZNK15QAbstractSocket10socketTypeEv + 318
8   QtNetwork                           0x000000010543f4ed _ZN12QLocalServer11qt_metacallEN11QMetaObject4CallEiPPv + 12013
9   QtWidgets                           0x0000000102ff560d _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent + 301
10  QtWidgets                           0x0000000102ff69b1 _ZN12QApplication6notifyEP7QObjectP6QEvent + 369
11  QtCore                              0x0000000105e16034 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent + 164
12  QtCore                              0x0000000105e68b4a _ZN27QEventDispatcherUNIXPrivate23activateSocketNotifiersEv + 250
13  QtCore                              0x0000000105e697b1 _ZN20QEventDispatcherUNIX13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE + 1169
14  QtCore                              0x0000000105e12341 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE + 417
15  QtCore                              0x0000000105c539d3 _ZN7QThread4execEv + 115
16  QtCore                              0x0000000105c573da _ZN18QThreadStorageData6finishEPPv + 2458
17  libsystem_pthread.dylib             0x00007fffd8eca9af _pthread_body + 180
18  libsystem_pthread.dylib             0x00007fffd8eca8fb _pthread_body + 0
19  libsystem_pthread.dylib             0x00007fffd8eca101 thread_start + 13

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fffd8ddf442 __getdirentries64 + 10
1   libsystem_c.dylib             	0x00007fffd8d13a3a _readdir_unlocked$INODE64 + 133
2   libsystem_c.dylib             	0x00007fffd8d13ae9 readdir$INODE64 + 35
3   com.apple.CoreFoundation      	0x00007fffc30a4072 _CFIterateDirectory + 114
4   com.apple.CoreFoundation      	0x00007fffc30a37e8 _CFBundleGetBundleVersionForURL + 472
5   com.apple.CoreFoundation      	0x00007fffc30a25de _CFBundleCreate + 238
6   com.apple.HIToolbox           	0x00007fffc2683847 islPopulateKLPropertiesFromBundle + 180
7   com.apple.HIToolbox           	0x00007fffc2670b8e islPopulateInputSourceProperty + 3206
8   com.apple.HIToolbox           	0x00007fffc266fc9e islGetInputSourceProperty + 62
9   com.apple.HIToolbox           	0x00007fffc266fb99 TSMGetInputSourceProperty + 425
10  com.apple.HIToolbox           	0x00007fffc26833e0 _CreateKeyboardInputSourcesArray_Mutable + 540
11  com.apple.HIToolbox           	0x00007fffc268370d _CreateKeyboardInputSourcesArray + 17
12  com.apple.HIToolbox           	0x00007fffc268358f SectSelectableInputSourcesArrayFromCurrentState + 94
13  com.apple.HIToolbox           	0x00007fffc267ea6e MyActivateTSMDocument + 764
14  com.apple.CoreFoundation      	0x00007fffc3113de4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
15  com.apple.CoreFoundation      	0x00007fffc3113a73 __CFRunLoopDoTimer + 1075
16  com.apple.CoreFoundation      	0x00007fffc31135ca __CFRunLoopDoTimers + 298
17  com.apple.CoreFoundation      	0x00007fffc310afa1 __CFRunLoopRun + 2081
[... 485 lines cut ...]
===========                       =======  ======= 
TOTAL                                2.3G     1222 
TOTAL, minus reserved VM space       1.3G     1222 

Model: iMac12,1, BootROM IM121.0047.B29, 4 processors, Intel Core i5, 2,5 GHz, 20 GB, SMC 1.71f22
Graphics: AMD Radeon HD 6750M, AMD Radeon HD 6750M, PCIe, 512 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353737334448302D4348392020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353737334448302D4348392020
Memory Module: BANK 0/DIMM1, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D5341384758334D314131333333433920
[... cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x9A), Atheros 9380: 4.0.74.0-P2P
Bluetooth: Version 5.0.4f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: ST3500418AS, 500,11 GB
Serial ATA Device: OPTIARC DVD RW AD-5690H
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
USB Device: Card Reader
Thunderbolt Bus: iMac, Apple Inc., 25.1

atdotde, 2017-05-14:
The problem appears to be in the networking code (possibly when subsurface tries to find out if there is a newer version available). Could it be that your networking setup is special in any way (proxy for example)?

Paru78, 2017-05-14:
I don’t think that there is something special in my network setup. I don’t use a proxy and I just checked my router config and the network setting on my Mac. As far as I see, everything is normal.

Paru78, 2017-05-14:
But your right, the problem is somewhere in the network. If I disconnect from the internet, I am able to open Subsurface. If I connect again and start Subsurface; same issue: Crash.

Paru78, 2017-05-15:
Do you have any solution for me to get it running? I really like subsurface after my offline test and like to use in a stable way.

atdotde, 2017-05-15:
This is really difficult to figure out. I also have Sierra here and for me it works fine. We have to figure out what is different on your system that interferes. 

Are you using homebrew or mac ports? Is there any chance that you (possibly via those package managers) have conflicting versions of either Qt or some crypto (libssl comes to mind) installed that erroneously get loaded? 

If you are willing to invest some time, you could try to build Subsurface yourself and then run it in a debugger (get the sources via git and follow the instructions in the INSTALL file, which should be straight forward but a bit lengthy since you probably need to install Qt as well). 

Maybe @dirkhh has some better ideas?

Paru78, 2017-05-15:
Thank you Robert,
Actually I am no programmer or have any strange things installed (as far as I know). 
I will try to follow you recommendations and hope I can manage. I really like to get it work.

dirkhh, 2017-05-15:
As a non-developer building it from source may be a challenge...

Here's one idea to see if we can find out more remotely.

Open a Terminal window (Cmd-Space and type terminal)

type 

/Volumes/Subsurface-4.6.4.dmg/Subsurface.app/Contents/MacOS/Subsurface -v -v

and send us the output from that. Maybe it will tell us more about what's going wrong.

atdotde, 2017-05-15:
Another thing: Could you try a slightly older version from https://subsurface-divelog.org/downloads/ for example 4.6.2? If that works (and tells you that there is an updated version), select not not check for further updates, save and quit. Then try 4.6.4 again. Now it should not check upon startup and maybe that prevents the crash.

Paru78, 2017-05-15:
Strange: 4.5 is working. Every following version not. Even when I refuse an update to newer versions.

neolit123, 2017-09-23:
hi. @Paru78 did you get this working?

neolit123, 2017-10-11:
closing this due to inactivity.
please, re-open if see fit.

closed 2017-10-11 by neolit123