subsurface/subsurface #400: pressure displayed for ruler off by factor 10?

sjvudp, 2017-05-15:
In subsurface 4.6.4 running on Windows I saw the ruler displaying a pressure of 94bar for a depth difference of 32 meters. Isn't that way too large, or did I misunderstand the display? The help file does not explain the output (enhancement request!), so I had to guess.

atdotde, 2017-05-15:
That pressure difference is in you cylinder, not ambient pressure.

sjvudp, 2017-05-15:
>>> "Robert C. Helling" <notifications@github.com> schrieb am 15.05.2017 um 09:46
in Nachricht <Subsurface-divelog/subsurface/issues/400/301401119@github.com>:
> That pressure difference is in you cylinder, not ambient pressure.

OK, I should have really presented an example, because I still thing it's wrong. I'll try to give a better example tomorrow, hopefully. But maybe try to explain the numbers of the ruler in the manual while you wait ;-)

sjvudp, 2017-05-16:
I must admit that I failed to reproduce the issue. May fault that I didn't take a screenshot in time. You may close this.

closed 2017-05-27 by dirkhh