subsurface/subsurface #401: Translation fix for dive planner (German): "Aufstiege in der Deko anzeigen"

sjvudp, 2017-05-15:
For 4.6.4 I think the German translation "Aufstiege in der Deko anzeigen" should actually be "Aufstiege in die Deko anzeigen" (I can't explain German Grammar well, but the first is used when describing a location, while the other is used when describing a direction, like "Ich sitze in der Bar" vs. "Ich gehe in die Bar")...

atdotde, 2017-05-15:
It's good as it is. Trust me. The meaning is "show ascents during the decompression phase" not "show ascends to decompression phase" (which is what you are suggesting).

sjvudp, 2017-05-15:
>>> "Robert C. Helling" <notifications@github.com> schrieb am 15.05.2017 um 09:45
in Nachricht <Subsurface-divelog/subsurface/issues/401/301400775@github.com>:
> It's good as it is. Trust me. The meaning is "show ascents during the 
> decompression phase" not "show ascends to decompression phase" (which is what 
> you are suggesting).

Thanks for the response. I was trying to make sense of the numbers, and the only interpretation that made sense to me was the interpretation I suggested. Maybe I'll have to re-thing over it, hoping for final enlightenment.

closed 2017-05-15 by dirkhh

sjvudp, 2017-05-16:
Maybe part of the confusion is wording: Somehow "decompression" and "safety stop" were treated as synonyms by me. On the other hand every ascent is "decompression". When looking at a concrete example like this, the only difference is that once there are two lines with "5m", or just one line.
[image]
My point is: Independently of whether the ascent is for decompression or not, it should be displayed (in text representation).
The other thing is that the ascent and descent ratios would really enrich the table.
Summary: I still think the meaning of the label "Aufstiege in der Deko anzeigen" is not clear.

atdotde, 2017-05-16:
This is a matter of choice (hence it's a configurable option): You say, you want to see the ascents (so you should tick that box). Other people only want to see the stops since the ascents are implicit (you have a 9m and a 6m stop, then you probably have to ascent between those. And you do that with the standard ascent rate). The latter prefers a more compact representation of the plan since they want to take it with them in the water and want to reduce the information to the required minimum. 

Plan a dive that requires more than  a single stop. Then you will more clearly see the difference.