subsurface/subsurface #416: [Bug] Erroneous datapoints when importing from Hollis DG03

debShelton, 2017-06-03:
With some dives, when I import from my Hollis DG03 dive computer, there are what seems to be single sample errors in the data (e.g. saying my depth was 82 ft in a dive where the depth on either side was 30-something ft and the true max depth was less than 50 ft). These errors do not occur when using Hollis's software from https://www.hollis.com/support/software/

Csv export from subsurface and Hollis attached. Also logging of the import and a screenshot.

[image]

[filesHollisProblem.zip](https://github.com/Subsurface-divelog/subsurface/files/1049913/filesHollisProblem.zip)

mturkia, 2017-06-27:
I see some quirks on the Holis CSV file, so I would think that Subsurface is just displaying the data as it is.
- at 16 minutes 2610 -> 1600 -> 1595 -> 2605 PSI
- at 23 minutes similar sudden drop

And I would assume the other spikes are the same. Naturally we could do some sanity checking on the values and not display the spikes, but I am not certain which approach is better.

janmulder, 2017-06-27:
Coincidentally, I did some research in this issue yesterday. I took the subsurface.bin from the zip file and used the libdc simulator to fake import the dives into Subsurface.  6 (or better 7, with one of 2 minutes) dives in that data and in 2 of which strange depth spikes, both up and down. I did not check the pressure drops (as these are wireless transmitted during the dive, occasional drops can be expected).

I found out that already at parser level (oceanic_atom2_parser.c) during the import, the strange spikes are there (so that they show up on the display, or in an exported csv file is logical).

I think we should ask @jefdriesen to take a look to this ... it might be something libdivecomputer related.

glance-, 2017-06-27:
Subsurface have some sliding average for the cylinder pressure just to compensate for issues like this.

I don't think its the right choice to filter such data in libdivecomputer, rather filter/average it in the PT calculations in subsurface.

janmulder, 2017-06-27:
@glance- it is not the pressure drops I am "worried" about (and I definitely agree not to filter/average in libdc). It is the weird depth spikes. Those could be the cause of numerous reasons, of which a parser/data format error is just one.

glance-, 2017-06-27:
Ah, sorry for the noise then. On the other side, it looks like we're not doing the pressure averaging any longer.

@dirkhh Am I missing something, or is that something broken? From what I can remember we used to average that noise out, but I can't find the code =/

jefdriesen, 2017-06-27:
This is probably a bug in the libdivecomputer parser. Such depth spikes are typically the result of using to many (or too little) bits. I already had this bug on my todo list, but so far I didn't had time to look into it.

dirkhh, 2017-06-27:
I don't recall filtering out the pressure noise. @torvalds  usually doesn't look at github issues - ask him directly / mailing list.

neolit123, 2017-09-23:
@glance- @dirkhh, @jefdriesen @torvalds let me know if this is ok to close.