subsurface/subsurface #426: Mobile UI #2: default action for main screen button

Bocio, 2017-06-08:
This idea follows (https://github.com/Subsurface-divelog/subsurface/issues/425)

## New dive/Add function

Subsurface follows a peculiar approach with a central icon that irremediably covers the last dive displayed. IMHO making worthless the space it saved not using the action bar. The classic approach would layout the Add button in the right lower corner. There is a reason for that: if the last item has a short description you do not cover it. In other words: we write and read left to right so the most important info should be always on the left. Of course, in Arabic language most info are mirrored. IMHO our Add symbol lies just in the center like a kick in the ass. This is the classic approach:

[image]

I'm obsessed with defaults. It means less UI click/tap to complete a task.
What is the most common action on Subsurface main screen? Is it "add dive manually"? Are we sure? Once the BT download feature will be ready I guess that I will find myself using subsurface mobile to download data from my Petrel 99% of the time. So we could have a choice for that button function. Hence an alternate version would be:

[image]

_More to follow..._

Ghalr, 2017-06-08:
Thank you so much.
Please keep your hard work :)

dirkhh, 2017-06-19:
I agree that the default action will likely change...

dirkhh, 2017-06-28:
@Bocio does the latest version of Subsurface-mobile reasonably address this issue for you?
What about the button icon? To me the down arrow you have in your mockup was too close to "download from the internet" - which is very different from "download from local device"; but I'm not sure that I like the icon I picked any better...

Bocio, 2017-06-28:
Let me see the latest build and I 'll give a feesback

davide@mobile

janmulder, 2017-06-28:
http://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.6.4.300-arm.apk

dirkhh, 2017-06-28:
That doesn't have reachability disabled, not sure it have today's Kirigami.
I'll make new binaries the moment I find ten minutes to do so

⁣-- 
From my phone​

http://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.6.4.300-arm.apk

https://github.com/Subsurface-divelog/subsurface/issues/426#issuecomment-311744775

Bocio, 2017-06-28:
Leave that button.
I would only change button color choosing an accent color to be used later in other UI part. I understand the now there are other priorities.
I quickly searched for a dive computer icon and license apart I found only this ugly console :)

https://png.icons8.com/scuba-computer/win8/1600

Google material icons has only a watch that could resemble a DC.

https://material.io/icons/#ic_watch

I could start from the Google watch adding two arrows (download/upload) inside the screen or at the side

BTW
While playing with the new BT functionality I tried a download without a dive computer... 
See the screencast

https://youtu.be/s08fD9UeT0g

dirkhh, 2017-09-23:
I think this is all implemented? @Bocio ?
Downloads (with a gorgeous icon) is now the default...

Bocio, 2017-09-25:
Yesss

closed 2017-09-25 by neolit123