subsurface/subsurface #428: Mobile UI #4: Dive list different layouts and menu option

Bocio, 2017-06-08:
Previous idea here: 
[Mobile UI #1: general layout](https://github.com/Subsurface-divelog/subsurface/issues/425)
[Mobile UI #2: default action for main screen button](https://github.com/Subsurface-divelog/subsurface/issues/426)
[Mobile UI #3: Dive list item layout](https://github.com/Subsurface-divelog/subsurface/issues/427)

All original mockups made with moqups and are online at: (https://app.moqups.com/Bocio/3SSyFCoEwP)

## Default dive list and menu option
I show here an idea for the main menu option:

[image]

It has six features:
- default dive list
- photo dive list
- featured dive list
- refresh aka cloud syncronize
- sort by date (current implementation)
- sort by divesite (more on this later)

## Default dive list
We already saw the default dive list. It has a fancy icon indicating dive type.

[image]

## Photo dive list

@atdotde put a great effort on enabling photos management inside Subsurface and while it's something I don't care I see that there are a lot of people who care. So now we have one or more photos attached to one dive. We could leverage this feature to have a dive list with one photo as thumbnail, Who don't use photos and choose list/magazine view will get a default thumbnail/image. It could be Subsurface logo, or a nice picture for each dive type (an actual photo not an icon). I can give you some CC licensed photos to use. A reb diver, a oc diver, a free diving diver etc...
Here an example:

[image]

Featured dive list

Here I copied the idea from Wordpress and Feedly mobile app. Basically once you have one or more photo linked to a dive you could create one cover for each dive. Please note that our current scroll method would not be ok for this kind of feature. We would need a page by page scrolling (I will add an explicative video later). We could just pick only certain dive for this kind of visualization i.e. only dives with 5 start rating...
Again, if you don't have photos attached you will get a default picture as previous example.
Anyway this is an example of how it could be:

[image]

dirkhh, 2017-09-23:
This issue has gorgeous ideas in it. I'd love to see some of this be implemented. I don't think I have the skill (or time) to tackle this, but year, I love this.

Bocio, 2017-09-26:
> This issue has gorgeous ideas in it. I'd love to see some of this be implemented. I don't think I have the skill (or time) to tackle this, but year, I love this.

Yes, everything is pretty much explained. We need "just" a developer on it.

dirkhh, 2017-09-26:
and once again I'll just name tag @jbygdell since @neolit123 forgot to add him.
This may be the killer feature for Subsurface-mobile 3.0 as it needs more infrastructure on the git side and the desktop version (we need to define this "one picture per dive" and also show this somehow on the desktop to make things consistent and useful, I think)

Maybe after 4.7 and 2.0 are stable, ready on all platforms, and released, we could tackle that.

mikeller, 2026-10-17:
Hi @Bocio,

Same note as on the rest of this series: the mobile dive list and its menu options have been reworked multiple times since 2017. We're closing this as stale/superseded.

Please reopen if something from this proposal still applies to the current mobile dive list.

closed 2026-10-17 by mikeller