subsurface/subsurface #430: [idea] Mobile UI #5: sidebar

Bocio, 2017-06-08:
Previous idea here:
[Mobile UI #1: general layout](https://github.com/Subsurface-divelog/subsurface/issues/425)
[Mobile UI #2: default action for main screen button](https://github.com/Subsurface-divelog/subsurface/issues/426)
[Mobile UI #3: Dive list item layout](https://github.com/Subsurface-divelog/subsurface/issues/427)
[Mobile UI #4: Dive list different layouts and menu option](https://github.com/Subsurface-divelog/subsurface/issues/428)

All original mockups made with moqups and are online at: (https://app.moqups.com/Bocio/3SSyFCoEwP)

## Main sidebar

As you can see in previous images there is big mess with strings longer than available space.  Nevertheless I’m sure the English version is ok. IMHO items groups are not semantically correct.  Moreover we have huge truncated strings and a ugly corner at the base. Comparing Subsurface sidebar font with other apps I immediately note three things:

- Sidebar font is way too big respect all other apps.
- Sidebar width doesn’t take in account string length.
- Subsurface opens submenus  with a nice slide effect while other apps tend to avoid submenu or they use a simpler tree organization (as Android guidelines strongly suggest).

I did not spend time to organize items differently but I made an attempts to a more streamlined and Android compliant layout. I just changed some of them in a hurry.

[image]

The main (and key) difference here is the use of standard Android tree like submenu (> icon). It's very important because when opening a submenu you do not waste/occupy any extra horizontal space. Ok here I wasted a small extra space with fancy icons.
Look a this example:

[image]

Yes we have settings. Using a proper setting sheet we could clear the sidebar from the more obscure options.

## Sidebar header

Ok, I let my imagination run free here.

- user can use one of his photos to customise sidebar header
- A gravatar image is grabbed automagically matching the user email against gravatar WS. User name is taken from there also or we just take the left part of the email address he used to authenticate to the cloud.

_more to follow..._

Bocio, 2017-07-18:
Following this idea I continued to elaborate it.

This is our current navigation scheme.
[image]

IMHO it's clear that features and items where added at different stages bringing not semantically correct groups.  Hence my rework of sidebar and navigation:

- I organized menu items
- I added a settings page/panel
- I moved some menu items to the new setting page

Before diving into the sidebar items some comments on general layout.

## Title bar
Ok, we don't have Action Bar anymore (R.I.P.) but we a title bar that should have a proper height with a proper font. https://material.io/guidelines/layout/structure.html#structure-app-bar
Right now we have a thin thin thin font and a sort of breadcrumb mimicking desktop apps (i.e. Dive list > Dive Computer). This is a nightmare for localization because you have to write two menu levels.
The standard title bar has just the current function/panel title and an icon to go back to the previous screen. Moreover it has a proper title font with correct padding from borders. Hereafter how our title bar should appear based on function/page we are:

[image]

## Sidebar list
We have to use use a simpler tree organization (as Android guidelines strongly suggest). Here I used standard Android tree like submenu (> icon). It's very important because when opening a submenu you do not waste/occupy any extra horizontal space. You can see that we have a two menu level with the same horizontal space. I hope Kirigami has this kind of component implemented.

[image]
[image]

## Sidebar Items

### Dives or Dive list
Basically this is our home page. Nothing new here

### Synchronize
This is the ""manual sync with cloud. By default Subsurface-mobile runs offline and only syncs the dive list with cloud storage when explicily told to do so. Hence I moved it to main menu item because it's important.
### Dive management
Currently is called "Manage dives". I removed the "offline mode" because this is a real setting nothing more. 
I added (actually I moved) the obscure "GPS-tag dives" item that was under GPS menu. I had to read the user manual to understand its meaning si  I renamed it to a more clear "Apply GPS fixes". I moved it here because the object you are manipulating here are the dives. You are really updating your dive data so it deserves to stay here in the "dive management" menu item.
### GPS
I moved the "Clear GPS cache" to the settings page.
I moved the "GPS-tag dives" for the reasons explained above
I forgot to leave "Show GPS fixes". I realized my mistake while writing. I have to modify the mockup. Sorry. Mea culpa.

### Developer
Bye bye... Moved to settings page
### User manual
I renamed it "Help" to be compliant with Android guidelines
###About
Nothing new
###Settings
This is a new page. No more sub items like "Preferences" or "Cloud credentials", just a plain Settings page. As per [Android UX guidelines:](https://material.io/guidelines/patterns/settings.html)

_- All of an app’s settings should be reached via the “Settings” label. Do not use synonyms such as “Options” or “Preferences.”_
_- If side navigation such as a navigation drawer exists, include “Settings” below all other items (except Help & Feedback)._

I tried to follow latest guidelines on labels and settings placement/layout. Of course, in my mockup I added everything in one long page. In real application we will have scrolling to see all of them. **I forgot developer option as well...**

[image]

jbygdell, 2017-07-18:
Other than the vertical foldout of sub-menus, which are where Kirigami can be an issue, I see no issues on implementing this.

Bocio, 2017-07-18:
Let me know what do you think about new menu organization.

dirkhh, 2017-07-18:
So many things to comment on... which makes me think that this should have be broken down into smaller issues :-)

I like the idea of showing the account name / email as part of the header image of the global drawer.
Not sure, though, where we'd get the avatar for that person - but that would be neat.

Title bar: To understand this better, you need to look at the structural idea behind Kirigami. The model that Kirigami uses and that we follow is a "stack or list of pages" - and the title bar reflects that. You can easily use the title bar to switch between different pages; I do that all the time and find it quite useful. The question of "proper size, font, spacing" is interesting. I find the HUGE GINORMOUS OMG LOOK HOW BIG I AM title bar in typical Google Android apps wasteful - especially since we don't use it as a place to hold our action icons. So yes, the padding is slightly off right now, but size wise I'd rather have more vertical space for the content of my page.
And then you look at apps like Waze and you see a model much closer to what we do. Waze has no title bar at all and moves the interaction to the bottom of the screen and (surprise, surprise) a drawer that you can open very much the same way our global drawer works :-)

I'm ok with Dives / home icon
I like Synchronize with the matching icon
Dive management and GPS also both seem ok
Shouldn't download from DC use the cool new icon that we have? Or should we ditch that and use the icon you have there in the menu? Having two different ones seems wrong

Help, About, Settings for top level menu all sound good.

I find the settings page somewhat inconsistent.
How do I interact with the text "Clear all GPS fixes cached on device"?
Same for "Save detailed log" - that maybe becomes a switch (and no, I don't really care for the weird slider style switches that Android loves - I find it interesting that they have changed from version to version to version because clearly everyone hates whatever Google comes up with for those - and Samsung uses yet different visuals for that, IIRC)

What I don't like is the Theme selection... I'm always annoyed with things that are named Blue and Pink but that aren't blue or pink (or at least show the colors the way we do today).

Overall this is definitely a great step in the right direction. My comments are more about details, not about the overall layout.

And I think making those changes right now shouldn't be too invasive / hard to do, even though we are a bit late in the 2.0 cycle.

jbygdell, 2017-07-18:
My idea for the theme switching is a switch that toggles on/off for dark theme and then a slider with Blue in our blue color on one side and Pink in Pink color on the other. That way the user can select the accent color of choice and both will be available for light and dark theme. And they will also see what the accent colors look like.

Bocio, 2017-07-18:
> So many things to comment on... which makes me think that this should have be broken down into smaller issues :-)

I could do it. I still see an edit icon even on my first post so I guess I can edit it.

> I like the idea of showing the account name / email as part of the header image of the global drawer. Not sure, though, where we'd get the avatar for that person - but that would be neat.

I see that a lot of websites use Gravatar API (I guess). I registered there some years ago and occasionally I find my gravatar whenever I register to a new site. Even Github uses it because I did not choose my avatar.. When could then having a default avatar if there's nothing online.

Title bar: To understand this better, you need to look at the structural idea behind Kirigami. The model that Kirigami uses and that we follow is a "stack or list of pages" - and the title bar reflects that. You can easily use the title bar to switch between different pages; I do that all the time and find it quite useful. The question of "proper size, font, spacing" is interesting. I find the HUGE GINORMOUS OMG LOOK HOW BIG I AM title bar in typical Google Android apps wasteful - especially since we don't use it as a place to hold our action icons. So yes, the padding is slightly off right now, but size wise I'd rather have more vertical space for the content of my page.

I'm with you on this. Once we ditched the action bar I'm ok to use a slim title bar. I'm just saying that we should use a font weight similar to dive title and just leave the correct margin around it. On my devices the title/page is literally attached to the left border.
Regarding the breadcrumb it's really naive on a mobile app. Moreover on italian localization I strive to find correct short words to accomodate two menu levels and I think this could be true in other languages too. 
Moreover if you look at my organization idea, the upper level is redundant. I don't know the kirigami stack/list of page but I hope you can only display the last leaf/page of the tree.

> And then you look at apps like Waze and you see a model much closer to what we do. Waze has no title bar at all and moves the interaction to the bottom of the screen and (surprise, surprise) a drawer that you can open very much the same way our global drawer works :-)

Yes I see but that is a complete different game. 
I remember Thomas commented that we could have up to three buttons in the bottom drawer. Now we use "download", we could think adding a "Search" button in a next release. But for now i think that a slim title bar that gives you the "context" is still useful. 

> Shouldn't download from DC use the cool new icon that we have? Or should we ditch that and use the icon you have there in the menu? Having two different ones seems wrong

Yes of course. I just used the first icon I found in the free online mockup app I'm using. I cannot upload icons in the free version.

> I find the settings page somewhat inconsistent.
> How do I interact with the text "Clear all GPS fixes cached on device"?

I searched through the Android tutorials and verified on some apps on my phone: you just press it and a confirmation dialog comes up. The first app that comes to my mind is Whatsapp. Go to settings page and try with clear/delete chat history or other cached data.

> Same for "Save detailed log" - that maybe becomes a switch (and no, I don't really care for the weird slider style switches that Android loves - I find it interesting that they have changed from [...]

That was my fault. Sorry.  I forgot the switch. It should be like offline mode setting.

> What I don't like is the Theme selection... I'm always annoyed with things that are named Blue and Pink but that aren't blue or pink (or at least show the colors the way we do today).

As I wrote I like the current implementation except for the radio button position. Drawing it on the mockup free app was a PITA so I just sketched a simplified version. BTW I see that Joakim has a new idea already.

willemferguson, 2017-07-18:
A few things.
The suggestions reflected in the images above constitute a large step in making the UI more efficient, streamlined and in line with other Android apps. Take my suggestions below within this context.
1) The menu in the sidebar will be getting much longer over the next 5 years or so, it's a given that comes with increased future functionality of Subsurface-mobile. Expansion of the menu needs to be designed in now so that re-design at a later stage is note required. There are already the beginnings of a dive planner on mobile that will need to be added to the menu system.
2) The home button probably requires the label "Dive List", much more specific and accurate than "Dive"
3) Dive log synchronisation should be under Dive Management. Under separate cover I will (again) make a case for "Download from cloud". The present synchronisation does not work as expected at all.

jbygdell, 2017-07-18:
Wilhelm, the planner could go under a "tools" section.
That way we are future proof if we decide to add something else.

willemferguson, 2017-07-18:
Joakim,
Just look at the items in the Desktop Main Menu, Things like "Filter", Renumber dive list, Edit device names, Configure DC, und so weiter und so weiter. Several of these items are likely to find their way into mobile and space needs to be available to build them into the menu system. Already I am not convinced that the whole menu system will fit onto a single mobile screen. Rather than an ad hoc solution such as "Tools", a more fundamental and pro-active plan needs to be made. Even if there is a Tools item in the present menu system (and Davide has forgotten at least one more menu item, Show GPS locations), the list is already too long for an average mobile phone.

willemferguson, 2017-07-18:
........ One can start with a decision that only one menu item can be expanded at a time, but I am not convinced that this will provide sufficient screen space for future expansion.

jbygdell, 2017-07-18:
A quick mockup on the reordering of the global drawer.

[image]

dirkhh, 2017-07-18:
Ok, so the mockup above looks MUCH better. I don't think we can easily move the arrow to the front and do the folding layout that @Bocio suggests, but the smaller font, tighter spacing, and especially the little icons... yeah, I'd love that.

jbygdell, 2017-07-18:
I don't think we can change the position of the arrow either, seems like kirigamis actions are not designed for that sort of fold out display. What we can do is to make sure that we only get one level of sub-menu items.

dirkhh, 2017-07-18:
and we should be able to make the font smaller... and I'm pretty sure that Actions already support the notion of an icon, right?

Bocio, 2017-07-18:
Ok better than nothing.

Thank you

jbygdell, 2017-07-18:
Dirk, yes on the icons, not sure about the text size. At least it will not accept font.pointSize definitions.

[image]
.

Bocio, 2017-07-19:
> So many things to comment on... which makes me think that this should have be broken down into smaller issues :-)

Done. I moved to different issues: sidebar header, title bar, settings page.

dirkhh, 2017-07-19:
Thank you for all the effort that you spend on improving the UI.
I just sent email to the mailing list (responding to Willem - "Subsurface mobile: delete and sync operations") which contains some interesting user experience questions that I'd love your thoughts on.

jbygdell, 2017-07-19:
Progress, developer menu can be hidden using a checkbox on the settings page.

[image]

dirkhh, 2017-07-19:
I like the icons. I'll try to figure out what we can do about the text size.

Bocio, 2017-07-19:
> Progress, developer menu can be hidden using a checkbox on the settings page.

Wow, it's taking a new shape!
IMHO by default developer menu should be disabled as on Android phones.

jbygdell, 2017-07-19:
@Bocio of course, what i really meant was, in order to display the developer menu you must activate a checkbox on the settings page.

jbygdell, 2017-07-19:
Latest set of patches pushed to MobileUI branch, need to fix the preferences page before I submit the PR. Might have to do something about the cloud manual sync and auto sync icons as well.

[image]

Bocio, 2017-07-19:
You are my hero

dirkhh, 2017-09-23:
I think this is mostly done... I read through it one more time but can't find anything else that is missing. @Bocio -- should this be closed? Or can we create a new issue for things still missing?

Bocio, 2017-09-25:
I Will give give you feedback tomorrow morning CET

davide@mobile

Bocio, 2017-09-26:
@dirkhh everything is done here.

Really thank you!

closed 2017-09-26 by Bocio

dirkhh, 2017-09-26:
Oh, the thanks go to you, @Bocio for the great ideas and suggestions and for your persistence in promoting them and to @jbygdell for the outstanding work on implementing them