subsurface/subsurface #440: IQ Dive Computer Support

jarango, 2017-06-15:
Wondering if you could add support for IQ Dive Computer (diveiq.com)? Its a Dive Computer app for Garmin watches. It saves dives as a normal activity such as running or cycling, but includes dive related information such as depth profile, no decompression time, max depth etc. These activities get written to the  watch as FIT files that can either be read through USB or download from Garmin Connect website once the watch has been synced with the Garmin Connect service. The official site of the FIT file format is https://www.thisisant.com/resources/fit. They have SDKs available for decoding the files. Here is an example of a dive. Scroll down and you will see multiple graphs related to diving.

https://connect.garmin.com/modern/activity/1796279118

One cool thing is that the location of the dive site is included in the data because the watch has GPS.  I am the developer of the app and I am willing to help you guys out if you decide to add support for this app.

Thanks
Jesus

neolit123, 2017-09-23:
@mturkia, @jarango 

jarango, sorry for the late reply. can you share one of files that the IQ Dive Computer exports. maybe it's possible to add support for it in Subsurface?

@mturkia not sure if @jarango will reply so do you think this is doable?
please, close this issue if see fit.

thanks

neolit123, 2017-10-11:
closing this due to inactivity.
please, re-open if see fit.

cmair, 2021-08-24:
I know this is quite an old issue, but I recorded quite a few dives with IQ Dive Computer and I'd love to see subsurface support as I'd like to finally bring my dive logs into the digital space. I can offer to share FIT files as needed and uploaded an example here: https://github.com/cmair/diving-files/blob/master/7255731073_ACTIVITY.fit
In case you need more recordings, just let me know!

dirkhh, 2021-08-24:
Could you share as much data as possible about this specific dive? What's the EXACT start time, according to the IQ? How long was the dive, how deep? What other information does the IQ show us?
What's the firmware version / serial number of your dive computer?

We have a parser for Garmin FIT files, but right now it seems to create mostly garbage when trying to parse the sample file - but the more we know what SHOULD be there, the easier it might be to piece together rules how to make sense of the file.

cmair, 2021-08-24:
Here is how Garmin handles the file: https://connect.garmin.com/modern/activity/7255731073
There you should be able to see all the information I could provide about the file.

The dive log on the watch shows me:
dive no: 84
time of entry: 13:23 on August, 6th 2021
max depth: 39.47m
duration: 47:57 (min:sec)

Speaking of times: the recording is much longer than the dive itself, as the watch records data such as GPS between manual start/stop points. Looking at the graphs, the dive starts approximately 16:12 minutes into the recording. According to https://www.fitfileviewer.com the activity was created at 13:06:46, so the 16 minutes offset comes close to what the watch reports as time of entry.

All other entries from fitfileviewer seem kind of valid to me: Session record with max depth (39.47489m) at the very bottom and records containing Depth, Ascent Rate, Ceiling, No Decompression Time and Gradient Factor together with lots of other information that gets recorded by default. The units of measurement can be changed from within the application on the watch, therefore I assume the numbers would look different if I had chosen imperial measurement units. There is also a "field_description" record showing those.

I think that's all the information I'm able to provide. I hope it helps a little. In any case, don't hesitate to ask for more, I'll try to help!

mikeller, 2026-10-18:
Hi @jarango,

cmair's detailed 2021 follow-up (session/record field structure, units-of-measurement caveat, GPS-tracking preamble timing offset) looks like it's still usable, the sample FIT file they linked is still accessible. Rather than asking for fresh data from scratch, can we confirm: is this still wanted, and would you (or cmair, if still around) be able to help verify that sample data is still representative of the app's current FIT output before someone picks up the reverse-engineering work?

cmair, 2026-10-24:
Hi @jarango,

I'm still around. Unfortunately I'm not using IQ Dive Computer anymore so I won't be able to provide new FIT files. I might be able to dig up a few old ones if that helps.

mikeller, 2026-10-25:
Thanks for getting back to us, @cmair. No need to dig up more files just yet. Given that you no longer use the app, I don't think we should prioritise support for its current output without someone who can test it.

Importing older IQ Dive Computer logs could still be useful. The sample FIT file linked above now returns 404, though, so if you would like to pursue historical imports and can restore that sample or share a few old files, we could revisit it with their expected dive times and depths. For now, I suggest we leave this as a low-priority historical import request rather than assume the app is still actively supported.