subsurface/subsurface #443: Bluetooth disapear

marccollin, 2017-06-19:
I use a petrel 2, habitually when i go in the download menu, I see bluetooth bottom 

but now it seem to have disapear.

[image]

i use 4.6.4

glance-, 2017-06-19:
Thats probably because you're build of subsurface is done without `BT_SUPPORT` . Where did your binaries come from?

marccollin, 2017-06-20:
suse

https://build.opensuse.org/package/show/home:Subsurface-Divelog/Subsurface

dirkhh, 2017-06-20:
The distro specific packages should support Bluetooth. I'll need to figure out what went wrong there... 

Guillaume, I'm traveling today (flying back from China), any chance you could take a look?

Thanks

/D

⁣-- 
From my phone​

https://build.opensuse.org/package/show/home:Subsurface-Divelog/Subsurface

https://github.com/Subsurface-divelog/subsurface/issues/443#issuecomment-309618009

ggardet, 2017-06-20:
@marccollin which openSUSE version ? And which Subsurface version ?
Here, I have openSUSE Leap 42.2 and subsurface 4.6.4 and I have the bluetooth checkbox.

marccollin, 2017-06-20:
hi, 

tumbleweed
4.6.4

ggardet, 2017-06-20:
Indeed, the build log says: 
`[  612s] -- Turning off Bluetooth support as Qt version .. is insufficient for that`
I will try to have a look at it soon.

marccollin, 2017-06-20:
no problem, will do my rebreather course in the next few day :)

ggardet, 2017-06-20:
@marccollin, the problem should be now fixed. Please update your subsurface version once the update will show up in your update software and tell me if it is now ok.

marccollin, 2017-06-21:
work fine now

another issue i saw but seem to exist since a few version... when i close
subsurface i get a crash.

ggardet, 2017-06-21:
@marccollin, since this is another bug, please open a new issue on github, please.

neolit123, 2017-09-23:
@ggardet @marccollin the other bug is covered in https://github.com/Subsurface-divelog/subsurface/issues/510

closing this issue.

closed 2017-09-23 by neolit123