subsurface/subsurface #454: Subsurface-Mobile: Problems with autosync

willemferguson, 2017-06-24:
1) Is it possible to do disable autosync when dragging the dive list down in the case where auto cloud sync has been disabled in the side menu for 'Manage dives'.?

2) When dragging the dive list down and triggering autosync, a blue circle appears onscreen. It looks like it could be an attempt towards a busy indicator, but it is not clearly recognisable as such. When autosync is done, the circular graphic does not disappear. See attached image.

[image]

willemferguson, 2017-06-24:
The only way to get rid of the blue circle is to drag the list upwards until the circle disappears. When pulled own again, the circle is gone.

janmulder, 2017-06-24:
Issue 1: I am pretty sure things are a little buggy here. When I switch to offline mode, I can still force download by pulling down, and I can use "manual sync with cloud". Pulling down sync is annoying, but "manual sync with cloud" is kind of ok, as it is a very deliberate act. I think it is annoying because when browsing trough recent dives, I like to be able to return to the top of the divelist using a fast swipe, and that will always trip a pull down sync. Functionally, the solution is simple ... disable the pull down sync when in offline mode.

As a side note. I was very confused when installing version -273 (self build). Pull down sync was not working at all. The app-log showed:  "but user asked not to connect to network". So I was in proper offline mode. After messing with "offline mode" on and off, I am not able to reproduce this behavior any more.

willemferguson, 2017-06-24:
See this screencast:
https://www.dropbox.com/s/47m9ujbzancuxen/2017_06_24_11_28_30.mp4?dl=0
This is on a Samsung S6 running Android V6.01. What I did here, was to drag the dive list down once. The response is jerky, with the list coming down, going up to the top of the screen and then coming down again. This is not dependent on any action of mine after initially drawing down the dive list.

Bocio, 2017-06-27:
@willemferguson as @janmulder already noted and referenced, we are basically speaking of the same problem #456.
I saw your screencast. Your problem is even worse because you have a huge dive trip as last element in the dive list. The UI hangs and the spinning circle stops the animation. So it's the third point I commented here https://github.com/Subsurface-divelog/subsurface/pull/456#issuecomment-311312178