subsurface/subsurface #457: Worldmap export / DeletedApiProjectMapError

elfrinjo, 2017-06-25:
Subsurface Version: 4.6.4-1 (Arch AUR Build)
Browser: Chromium 59.0.3071.109 (and others)

Chromium show the following Java Script errors:

```
Google Maps API error: DeletedApiProjectMapError https://developers.google.com/maps/documentation/javascript/error-messages#deleted-api-project-map-error
_.rb @ js:35

maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:28 Uncaught TypeError: Cannot read property 'fromLatLngToPoint' of undefined
    at IU (maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:28)
    at _.td.KU.b.b (maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:26)
    at Object._.wd (js:57)
    at maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:30
    at js:50
    at maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:30
    at maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/marker.js:50
    at Az.B (maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/util.js:207)
    at maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/util.js:208
    at maps.googleapis.com/maps-api-v3/api/js/29/7/intl/de_ALL/util.js:212
```

mturkia, 2017-07-24:
Our API key has apparently been deleted by Google. I have no idea why this has been done, but a valid API key is needed for the map to function. Individual users should be able to create their own API key and update the exported worldmap with that, but I am not sure what the licensing terms are for open source projects.

IIRC we concluded that this kind of use is OK when the feature was implemented, but as our key has been deleted, someone should re-check the licensing terms before we get a new API key. Or we should consider options for the user to insert their own key when exporting the worldmap.

neolit123, 2017-09-23:
this should be resolved by 5b28fcea3a731.

closed 2017-09-23 by neolit123