subsurface/subsurface #471: [idea] desktop map view with dive clusters

Bocio, 2017-07-07:
Sample wireframe:

http://i.imgur.com/o8IP4Ho.png

I took this idea form an online logbook (diveaboard.com) and I saw this bubbles in other google maps based projects. (a local newspaper here in Italy use the same bubbles to display news location).
Basically it gives you the concentration of your dives. Each color bubble indicate/aggregates a number-of-dives range. Bubbles scales according to the map zoom. 

**Navigation idea:**

Once you zoom-in after a certain scale you get a single dive spot view like our current Marble map.
We are not editing dives here so it would be nice that once I hoover on a given dive spot I get a small box with main dive details.
Inside this small box I could have a link/button to the real dive info.
Of course, once I click I will abandon the Map View and I go to our classic view.

As @neolit123 wrote this is possible with MarkerCluster library of Google Maps API:

https://developers.google.com/maps/documentation/javascript/marker-clustering
https://github.com/googlemaps/v3-utility-library/tree/master/markerclusterer

He even sketched the cluster idea. Here is a preview:

https://www.dropbox.com/s/6gdygq68uba2pg8/cluster_clicks.png?dl=0

Each dive/marker now now has max. depth value. by clicking a cluster on the map at the bottom of the Qt UI it says "selected a cluster with N dives and average depth M". So basically stats are possible for a dive cluster.

neolit123, 2017-09-23:
@Bocio 
Davide, i will rest this for a while without closing it but after the latest googlemaps widget that we have:
- the complexity of clusters implementation has raised a lot, because there are big limitations on the QtLocation Models side (the new map framework that we use). not as easy to do as in the "native" googlemaps.
- we already have similar functionality in terms of selecting multiple nearby dives by clicking a single marker, so it's a matter of showing a special cluster icon which is shown when the zoom level is too high and the markers should then be hidden and that's the hard part.

if QtLocation decide to add cluster support we can possibly add it in Subsurface too, but until then this is a change-request which is not worth it.

Bocio, 2017-09-26:
@neolit123 I agree with you. 

I opened this as a remainder of how this feature could be implemented but, as you wrote, we have a good compromise with the current map framework.
I close this.

Thanks!

closed 2017-09-26 by Bocio