subsurface/subsurface #478: Adding Pictures on Tab "Bilder" fails

Natas-Lived, 2017-07-12:
On the tab "Bilder", space is reserved for pictures. Right-clicking anywhere in this area opens a context menu "Bild(er) aus Datei(en) laden / Bild(er) aus Web laden..." but selecting any of the entries doesn't result in any action. There is no upload dialogue.
When selecting an entry of the list of dives, right-clicking on it, there is a context menu offering "Bild(er) aus Datei(en) laden". Selecting this opens a dialogue and pictures can be selected and loaded.

Expected:
Using the context menu on the "Bilder" tab leads to a upload dialogue.

dirkhh, 2017-07-12:
Fascinating. I didn't even recall that we had this feature. And clearly it hasn't been tested in a while since it's clearly broken.

As a work around, right click on the dive in the dive list - you'll see a longer context menu which also contains "Bild(er) aus Datei(en) laden / Bild(er) aus Web laden..." - those two options appear to still work. In the meantime I need to figure out why this doesn't work when triggered from the pictures tab...
Hey @tcanabrava - wanna take a look?

tcanabrava, 2017-07-12:
Looking

dirkhh, 2017-07-12:
too late... already fixed it

commit 1b8b1120 (2017-07-12): Fix slot handling in Photos tab

This clearly has never been tested, never worked.
Oh well.

Fixes #478
closed 2017-07-12 by dirkhh