subsurface/subsurface #479: SAC wrongly calculated?

Sm0q, 2017-07-12:
Hello,
maybe I do something wrong, however it seems that SAC is wrongly calculated in my case. You can see in the screenshots configuration (double tanks 2x12l, start/end air pressure - 150->50bar and avg. depth - 15.7m). According to my simple calculations, SAC should be ~23l/min while it Subsurface shows 5l/min. 
I wish it is true, however I it's waay to low value.
[image]
[image]

atdotde, 2017-07-12:
Could you please Export that dive as a subsurface xml and Mail it to me (helling@atdotde.de). Then I'll have a look tonight. 

Thanks
Robert

Sm0q, 2017-07-13:
Hello Robert,
I sent you logs as requested. I also realized that this may be an issue with Ratio computers log import. 
I recently bought iX3M Deep and all dives imported from this computer show incredibly low SAC values. I also exported 1 additional log (from GENZ Black EANx computer) as a reference (which shows much more realistic SAC - 18l/s). 
So again - in export you'll find 3 dive logs (RATIO iX3M), SAC < 5l/min and 1 log (GENZ Black) with SAC ~18l/s.

I also use (due to Ratio import feature in libdivecomputer) development version of Subsurface (compiled under Linux)

Thanks for help :)

atdotde, 2017-07-13:
Thank. I found the problem (or at least: part of it): All the problematic dives have a surface pressure of more than 9bar in the xml which totally screws up the average depth calculation (or better the conversion of average depth to average pressure). There must be something wrong with the import of that value from the dive computer. Until that is fixed, there is an easy workaround: Open the xml file of your log (if you use the cloud export it as xml) and manually delete all lines with ridiculous surface pressures. That makes  Subsurface assume a surface pressure of 1013mbar and you get reasonable SAC rates.

Sm0q, 2017-07-13:
That was a quick one - thanks :)
I'll also adress this to libdivecomputer, as they did recently some changes due to changed Ratio log format. 
Anyway - thanks again for your time :)

atdotde, 2017-07-13:
I already reported this to Jef, the owner of libdivecomputer.

Sm0q, 2017-07-13:
Ok, thank you. Closing this as not related to Subsurface itself.
Regards,
Jerzy

closed 2017-07-13 by Sm0q