subsurface/subsurface #481: password restrictions - correctly communicate

dirkhh, 2017-07-13:
Apparently we are silently failing on -mobile, claiming failed connection instead.

bluepanter, 2017-07-18:
Probably it is the TODO in /core/cloudstorage.cpp on line 74-76 in function CloudStorageAuthenticate::uploadFinished()

> 		```
/* TODO: Emit a signal with the error
		MainWindow::instance()->getNotificationWidget()->showNotification(get_error_string(), KMessageWidget::Error);
		*/
```

Ghalr, 2017-07-28:
Is it fixed now ?

mturkia, 2017-07-28:
> On 28 Jul 2017, at 7.51, Ghalr <notifications@github.com> wrote:
> 
> Is it fixed now ?
> 
The error message is now better, but I did not add anything to instruct about allowed characters.

neolit123, 2017-10-07:
closing this as `QMLManager::saveCloudCredentials()` already instructs the user about the allowed password character when the user is saving his credentials.

closed 2017-10-07 by neolit123