subsurface/subsurface #490: QML style download screen

janmulder, 2017-07-19:
I see some issues in the newly styled download screen:
1) reference to `isBluetooth.leftPadding` ... the isBooltooth checkbox is gone, so logcat warnings.
2) 3 screenshots below. 1st: after download all looks fine. 2nd: `unselect all` button pressed, still looks fine. 3rd: select all: blue background appears (theme is pink). 

[image]
[image]
[image]

3) I would like same line height as in divelist (measured on my screen 9mm in divelist, almost 12mm in download list).

janmulder, 2017-07-19:
Propose to close this issue as nr 1/2 are fixed by referenced commit. And open a new one for 3), which seems beyond my QML skills :-)

dirkhh, 2017-07-19:
Your commit closed this - let's re-open for the third issue. I agree, the lines are way too tall

janmulder, 2017-08-07:
fixed in 8c0a1c2e6d357a6

PR #494 (2017-07-19): QML UI: fix some obvious UI errors
Fixes: #490 items 1) and 2).

1) Reference to `isBluetooth.leftPadding` is removed as the `isBluetooth` checkbox is gone.

2) Do not set background color of downloaded dives. The checkbox is used to denote "selected or not".

In addition, incorrect references to `subsurfaceTheme.PrimaryColor` (which does not exist) are replaced by the correct `subsurfaceTheme.primaryColor`.

Finally, an obvious copy/paste error `width: childrenRect.height`. that is supposed to be `width: childrenRect.width`.
commit 6f8e50f8 (2017-07-19): QML UI: fix some obvious UI errors

Fixes: #490 items 1) and 2).

1) Reference to isBluetooth.leftPadding is removed as the isBooltooth
checkbox is gone.

2) Do not set background color of downloaded dives. The checkbox is used
to denote "selected or not".

In addition, incorrect references to subsurfaceTheme.PrimaryColor (which
does not exist) are replaced by the correct subsurfaceTheme.primaryColor.

Finally, an obvious copy/paste error width: childrenRect.height. that
is supposed to be width: childrenRect.width.
closed 2017-08-07 by janmulder