subsurface/subsurface #491: [idea] Mobile UI #8: Sidebar Header

Bocio, 2017-07-19:
Previous idea here:
[Mobile UI #1: general layout](https://github.com/Subsurface-divelog/subsurface/issues/425)
[Mobile UI #2: default action for main screen button](https://github.com/Subsurface-divelog/subsurface/issues/426)
[Mobile UI #3: Dive list item layout](https://github.com/Subsurface-divelog/subsurface/issues/427)
[Mobile UI #4: Dive list different layouts and menu option](https://github.com/Subsurface-divelog/subsurface/issues/428)
[Mobile UI #5: Sidebar](https://github.com/Subsurface-divelog/subsurface/issues/430)
[Mobile UI #6: Dive detail title](https://github.com/Subsurface-divelog/subsurface/issues/431)
[Mobile UI #7: Dive list EOL and Swipe To Refresh](https://github.com/Subsurface-divelog/subsurface/issues/432)

All original mockups made with moqups and are online at: (https://app.moqups.com/Bocio/3SSyFCoEwP)

## Sidebar header

[image]

- User can use one of his photos to customise sidebar header background. We could let the the user choose an image form its phone image gallery.
- A gravatar image is grabbed automagically matching the cloud user email against gravatar WS. Gravatar API is pretty simple and it not requires authentication: https://en.gravatar.com/site/implement/
If no image is retrieved from Gravatar we can specify a default image to be used directly into the http request.
- User name is taken from there or we just take the left part of the email address he used to authenticate to the cloud.

neolit123, 2017-09-23:
@dirkhh @janmulder

Bocio, 2017-09-26:
Implementing just the Gravatar API (like here on Github) would be a great improvement.

dirkhh, 2017-09-26:
I really like that, TBH. @jbygdell -- don't know if you have time for any of this right now, though. I don't think I'll have time for it any time soon.

jbygdell, 2017-09-27:
Not me either, and personally I think fixing the dive edit takes priority.

dirkhh, 2017-09-27:
Yes. Key concerns for Subsurface-mobile 2.0:
- cloud credentials
- dive edit
After that I'll release 2.0, I think.
The gravatar and other cool ideas from @Bocio can then be the new features for 2.1 and later

mikeller, 2026-10-17:
Hi @Bocio,

Same note as on the rest of this series: the mobile sidebar has been reworked multiple times since 2017 as part of broader UI rebuilds. We're closing this as stale/superseded.

Please reopen if something from this proposal still applies to the current mobile sidebar.

closed 2026-10-17 by mikeller