subsurface/subsurface #492: Mobile UI #9: Settings page

Bocio, 2017-07-19:
Previous idea here:
[Mobile UI #1: general layout](https://github.com/Subsurface-divelog/subsurface/issues/425)
[Mobile UI #2: default action for main screen button](https://github.com/Subsurface-divelog/subsurface/issues/426)
[Mobile UI #3: Dive list item layout](https://github.com/Subsurface-divelog/subsurface/issues/427)
[Mobile UI #4: Dive list different layouts and menu option](https://github.com/Subsurface-divelog/subsurface/issues/428)
[Mobile UI #5: Sidebar](https://github.com/Subsurface-divelog/subsurface/issues/430)
[Mobile UI #6: Dive detail title](https://github.com/Subsurface-divelog/subsurface/issues/431)
[Mobile UI #7: Dive list EOL and Swipe To Refresh](https://github.com/Subsurface-divelog/subsurface/issues/432)
[Mobile UI #8: Sidebar Header](https://github.com/Subsurface-divelog/subsurface/issues/491)

All original mockups made with moqups and are online at: (https://app.moqups.com/Bocio/3SSyFCoEwP)

## Settings

[image]

This is a new page. No more sub items like "Preferences" or "Cloud credentials", just a plain Settings page. As per [Android UX guidelines:](https://material.io/guidelines/patterns/settings.html)

_- All of an app’s settings should be reached via the “Settings” label. Do not use synonyms such as “Options” or “Preferences.”_
_- If side navigation such as a navigation drawer exists, include “Settings” below all other items (except Help & Feedback)._

I tried to follow latest guidelines on labels and settings placement/layout. Of course, in my mockup I added everything in one long page. In real application we will have scrolling to see all of them. **I forgot developer option as well...**.
I did not have the time to draw the current theme switcher that is very nice. Just for consistency radio buttons should go on the right side as other controls.

RickMWalsh, 2017-07-19:
I like it. Working on it now. Should have a pull request tonight my time

Bocio, 2017-07-19:
Hold on!

Look at @jbygdell comment https://github.com/Subsurface-divelog/subsurface/issues/430#issuecomment-316492072

Maybe he is already on it

RickMWalsh, 2017-07-19:
> Hold on!
>
> Look at @jbygdell <https://github.com/jbygdell> comment #430 (comment)
>
> Maybe he is already on it

Wow, looks great. I'll pull Jocke's changes when I finish work and see if
there's more I can do.

jbygdell, 2017-07-20:
@RickMWalsh I have a set of uncommitted changes as well for the settings page.

RickMWalsh, 2017-07-20:
@jbygdell 
I've pushed what I did (consolidated preferences and cloud into settings, made things align, and replaced the check boxes with styled switches) to github
https://github.com/RickMWalsh/subsurface/tree/settingsForJocke
Take as much or as little of it as you like.

jbygdell, 2017-07-20:
Nice, I had to do some creative rebasing in order to get it to merge in the end.

mturkia, 2017-07-24:
Currently changing the user credentials on settings page is broken. It feels like there is no action occurring when setting them, and there is no indication whether credentials are correct or not.

Anyway, one can invalidate the credentials in settings page and when going to divelist the good old cloud credentials page is shown where the user is given some indication of what is occurring.

So what should it be for setting cloud credentials, fixing the settings page so everything working smoothly there, or the old credentials page that currently works reasonably well for?

Bocio, 2017-07-24:
What do you mean invalidate the credential?

Yes it's broken a now. As quick solution we could just open the old auth panel from the settings page. I would not remove it from setting. Its really a one time operation for most users and it doesn't deserve to stay in the menu. Even on desktop it's into prefs.
The problem is that both desktop and mobile credential screens are broken because that are stateless while the registration is a two step process.

mturkia, 2017-07-24:
The only way to change to different account was insert incorrect creds and try to go to divelist. this brought in the old dialog.

The quick fix sounds good, but then again if you a proper fix planned, I can certainly wait.

Bocio, 2017-07-24:
I did not touch that part because I know Dirk manages part of it manually.

I'll try to decompose it into the different use cases. We'll see if we come out with something better

dirkhh, 2017-07-25:
I think there are a couple of different directions we could take it.
I'm not sure this is the best one, but this is one that would be consistent. Instead of using the Action button, have a button on the (sub-) page that walks you through the steps and changes as you go along (and changes the page with it)

starting point: just a field with the email address
two buttons below it "Use as cloud storage credentials" / "No cloud"

once you enter an email address and tap the left button you either get an error if there are invalid characters, or you get a second, protected field that you can enter a password in
one button "Save password"

if the password is accepted, add the text <verified> behind the email address, remove the password field
if not, give error
if PIN is needed, add the third input field an matching button

OK, as far as outlining all the possible cases this sucked. But I think you get the idea. This could be by itself on the cloud credentials page, or as a sub-page for settings.

Bocio, 2017-07-25:
Unfortunately I cannot work on this for the next two days I think. My job call me.

In the meantime I was thinking to the thorny problem of action bar and action button... Maybe for Subsurface mobile 3.0 :-)

- you dislike action bar because it wastes a lot of screen estate and it's far form thumb reachability.
- we will need several accessible quick actions without the needs of opening the sidebar.

Look at the DU recorder upper bar:

[image]

- No action button but several small icons with quick actions. Just imagine to have this in the bottom part instead of the huge action button (that part is already unusable because the center action button already cover most of info behind).
- We could remove the title bar: our menu structure and info displayed are very straightforward so there's no need to help the user understanding where he is.
- We could just add the hamburger icon on the left to open the sidebar. Then we would have: "Download from DC"; "Add new dive"; "Start GPS"and maybe "Search".
- Some icons (as you seems to suggest above) could change based on the context we are in. i.e. if you are on "dive details" panel you have the "edit" icon instead of "add new dive" icon.

Just my thoughts to add more confusion :-p

Bocio, 2017-07-26:
Comments on user authentication management follows here: https://github.com/Subsurface-divelog/subsurface/issues/515

PeterZaal, 2017-10-15:
In Subsurface-mobile v2 the header for the time and distance settings for the GPS service is currently labeled 'Subsurface GPS data webservice'. I feel this is incorrect, because it's not about the webservice. Further more, the prefix Subsurface is unneccesary. I'd rather see a simple label like "GPS" (like in the mockup) or "GPS service".

jbygdell, 2018-09-21:
@dirkhh I think this can be considered done.

dirkhh, 2018-09-21:
agreed

closed 2018-09-21 by dirkhh