subsurface/subsurface #493: [idea] Mobile UI #10: Title Bar

Bocio, 2017-07-19:
Previous idea here:
[Mobile UI #1: general layout](https://github.com/Subsurface-divelog/subsurface/issues/425)
[Mobile UI #2: default action for main screen button](https://github.com/Subsurface-divelog/subsurface/issues/426)
[Mobile UI #3: Dive list item layout](https://github.com/Subsurface-divelog/subsurface/issues/427)
[Mobile UI #4: Dive list different layouts and menu option](https://github.com/Subsurface-divelog/subsurface/issues/428)
[Mobile UI #5: Sidebar](https://github.com/Subsurface-divelog/subsurface/issues/430)
[Mobile UI #6: Dive detail title](https://github.com/Subsurface-divelog/subsurface/issues/431)
[Mobile UI #7: Dive list EOL and Swipe To Refresh](https://github.com/Subsurface-divelog/subsurface/issues/432)
[Mobile UI #8: Sidebar Header](https://github.com/Subsurface-divelog/subsurface/issues/491)
[Mobile UI #9: Settings page](https://github.com/Subsurface-divelog/subsurface/issues/492)

All original mockups made with moqups and are online at: (https://app.moqups.com/Bocio/3SSyFCoEwP)

## Title Bar
Ok, we don't have Action Bar anymore (R.I.P.) but we have a title bar that should have a proper height with a proper font. https://material.io/guidelines/layout/structure.html#structure-app-bar
Right now we have a thin thin thin font and a sort of breadcrumb mimicking desktop apps (i.e. Dive list > Dive Computer). This is a nightmare for localization because you have to write two menu levels. Moreover with a new menu organization, top level menu is redundant.

- The standard title bar should be slim.
- It displays just the current function/panel title and an icon to go back to the previous screen. Moreover it has a proper title font with correct spacing from borders. Hereafter how our title bar should appear based on function/page we are. Title bar height is just too big. It should be slimmer.

[image]

neolit123, 2017-09-23:
@janmulder @dirkhh marking this an an open-discussion.

Bocio, 2017-09-26:
IMHO this is a small mod with a big impact on app layout and usability. We don't need a breadcrumb with the new sidebar impl #430

dirkhh, 2017-09-26:
So the breadcrumbs are a feature of the underlying toolkit. Personally I find them somewhat useful, but we can certainly play with removing them. As usual, I have way too many things on my plate (things like "get BLE download to work at all on iOS" - which I would argue is in the big picture somewhat more pressing than this). @janmulder or @jbygdell , could one of you look into this one?

mikeller, 2026-10-17:
Hi @Bocio,

Same note as on the rest of this series: the mobile app's title bar and overall chrome have changed substantially across several UI rebuilds since 2017. We're closing this as stale/superseded.

Please reopen if something from this proposal still applies to the current mobile title bar.

closed 2026-10-17 by mikeller