subsurface/subsurface #497: [bug] Mobile app crashes deleting a dive

Bocio, 2017-07-21:
On Subsurface-mobile-4.6.4.443-arm.apk

I tried to remove some dives keeping pressed one dive from the dive list and tapping the bin icon. Each
time Subsurface crashed but the dive was deleted.

Screencast:

https://www.youtube.com/watch?v=NZcpPPI_Pyk

dirkhh, 2017-07-25:
I am able to recreate, but I haven't figured out how to fix it, yet.

Bocio, 2017-07-25:
I was not able to reproduce it on the latest apk but now everything is saved on the cloud.
In my test I was deleting dives just after downloading them from the Viper DC. 
I will try to clear everything and start with a new download later.

RickMWalsh, 2017-07-27:
I can reproduce it, but not consistently and I can't work out why it crashes sometimes but not other times.

PR #540 (2017-08-10): DiveList.qml: set detailsWindow index before deleting dive from list
This appears to fix the mystery crashes that can occur when deleting a dive from the dive list.

Fixes: #497
commit 1409521e (2017-08-10): DiveList.qml: set detailsWindow index before deleting dive from list

This appears to fix the mystery crashes that can occur when deleting a dive
from the dive list.

Fixes: #497
closed 2017-08-10 by dirkhh