subsurface/subsurface #507: Crash when downloading from Mares M1 dive computer

MartinKoeditz, 2017-07-25:
I'm download dive logs from Mares M1 via PC interface.
By reaching 100 % the application crashes.
Subsurface: 4.6.4-1
Environment: Ubuntu 16.04 x64

No crash if I check "save dump file of libdivecomputer".

Do you need some more information or dumps?
[Log und Dump.zip](https://github.com/Subsurface-divelog/subsurface/files/1173261/Log.und.Dump.zip)

janmulder, 2017-07-25:
I just correctly imported the dump file, using the libdc simulator (and a libdc compiled with `-DENABLE_PTY` to prevent tripping DTS/RTS serial line stuff due to using a virtual dive computer). So, basically, I can't reproduce the crash. 

Given the dates of the exact 50 downloaded dives (between 2000 and 2009), it seems that the Mares is not recently used. @MartinKoeditz: is this right? And are you helped with a  xml file of the dives that I downloaded?

MartinKoeditz, 2017-07-25:
Me and my wife just started diving. Each of us owns a Mares M1 now. So there are two new dives.

The described problem exists on both of our notebooks. Each running Ubuntu 16.04 x64. Maybe there are some side effects with the environment. I've installed the software from the "ppa:subsurface/subsurface" repository.

I can upload a crash report if that helps.

janmulder, 2017-07-26:
> I can upload a crash report if that helps.

Not sure it helps me, but maybe some of the other developers. So yes, please upload this.

Further, I do not see any recent dives from the dump file. Maybe the clock setting on the Mares is wrong (just a guess, as I do not have any Mares DC). What confuses me, is your remark "we have 2 Mares M1, so there are 2 new dives". The dump file, obviously, contains data from one M1.

With respect to environment. You might consider trying the AppImage (https://subsurface-divelog.org/downloads/Subsurface-4.6.4-x86_64.AppImage). See https://subsurface-divelog.org/download/ This might rule out any issue with the Ubuntu build.

MartinKoeditz, 2017-07-28:
> Further, I do not see any recent dives from the dump file. Maybe the clock setting on the Mares is wrong (just a guess, as I do not have any Mares DC). What confuses me, is your remark "we have 2 Mares M1, so there are 2 new dives". The dump file, obviously, contains data from one M1.

Yes, you're right. I didn't set the clock, so the Mares starts with year 2000. My wife set hers correct.

I added the Ubuntu crash dump. Maybe somebody can find a reason for the crashes.
[_usr_bin_subsurface.1000.crash.zip](https://github.com/Subsurface-divelog/subsurface/files/1183964/_usr_bin_subsurface.1000.crash.zip)

janmulder, 2017-07-28:
@ dirkhh: a suspicious  fragment in the crash report:

#0  strlen () at ../sysdeps/x86_64/strlen.S:106
 No locals.
 #1  0x00007f6a345b747e in __GI___strdup (s=0x0) at strdup.c:41
         len = <optimized out>
         new = <optimized out>
 #2  0x00000000005caaf2 in DownloadFromDCWidget::updateProgressBar() ()

Looking at updateProgressBar(), the free/strdup handling of the last_text variable looks a little risky to me. Possible free of something not allocated. Agree?

It is a bit of a long shot, as I can't reproduce the crash, but that is also a property of memory corruption issues.

Obviously, a change to be more safe here is trivial.

dirkhh, 2017-07-28:
would love a PR - I'm trying to get the no-marble builds out during lunch break

closed 2017-07-28 by dirkhh

janmulder, 2017-07-29:
@MartinKoeditz. As I'm not sure how familiar you are are with the Subsurface development model. There is a possible fix, but you will have to wait for it to show up in an official release (assuming that you want to stay with the Ubuntu ppa repo). Obviously, we are very interested whether is it truly fixed, so hoping to hear from you when the time comes.

commit b3da8847 (2015-03-15): Change the zoom level on globe if no coordinates are set

As flo269 suggested in bug #507, a full view of Earth would be better
when a dive has no coordinates.

Fixes #507

Reported-by: flo269
PR #519 (2017-07-28): Fix possible memory corruption
Version 2: 

A user reports a SIGSEGV that points to DownloadFromDCWidget::updateProgressBar() in relation to strlen()/strdup(), at end of download. Reading the code, as I can't reproduce the crash, it seems that the progress_bar_text is set to NULL and later strdup-ped. The man page is not fully clear on this, but setting it
to the empty strings is much safer.

Might fix: #507
commit 7e39be43 (2017-07-28): Fix possible memory corruption

A user reports a SIGSEGV that points to DownloadFromDCWidget::updateProgressBar()
in relation to strlen()/strdup(), at end of download. Reading the code, as I
can't reproduce the crash, it seems that the progress_bar_text is set to NULL
and later strdup-ped. The man page is not fully clear on this, but setting it
to the empty strings is much safer.

Might fix: #507