subsurface/subsurface #510: [Idea] Mobile: Double click on the back button to exit without confirmation box

Ghalr, 2017-07-26:


janmulder, 2017-07-26:
Can you provide a screenshot of the confirm box you like to see removed?

Ghalr, 2017-07-26:
I did not try the latest Android build. I suggest to use double-click than a confirmation box to exit the app. A lot of Android apps do use this behavior.

janmulder, 2017-07-26:
So what Android build did you try then? I understand what you suggest, but I have problems finding the corresponding code. So a screenshot would help with that.

Ghalr, 2017-07-26:
Trying the 4.6.4.508 build now

Ghalr, 2017-07-26:
OK to exit the latest build when you are at home page just 1-click to return button will result an exit. It will be better to double-click to prevent a wrong action.

janmulder, 2017-07-26:
So, then you better rephrase your request. There is no confirmation box (which I correctly could not find in the code). But you do want a double click end, and not the current single click ... 

I think you are a little confused here. A lot of android apps just close the screen on a single click back, leaving the app in memory which can immediately be opened from your recent apps. 

However, I am seeing a hard exit on 1-click back which is actually a crash. That crash is known for a long time, and is not in Subsurface code (but most likely Qt). Not that this matters to the average user,  but it matters to developers :-)

neolit123, 2017-09-23:
@janmulder was this resolved?
(you are going to have collaborator access soon - please close, if see fit)

neolit123, 2017-09-23:
renamed the title

janmulder, 2017-10-18:
This is most likely: https://bugreports.qt.io/browse/QTBUG-61434 , which is at this moment still in review (there seems a fix ready but not reviewed, or still under discussion).

Highly annoying bug, but we just have to wait for the Qt development process to do its thing. Closing it here.

closed 2017-10-18 by janmulder