subsurface/subsurface #512: [Bug] Mobile: Password is cleared with "show password" ON

Ghalr, 2017-07-26:
Hi.
Using fresh install of 4.6.4.508 build -
Type email + password (password is hidden)
Click now on "Show password"

-- A message is displayed "Opening local data file failed"
-- The password is cleared, the password field is now empty and we have to type the password again.

Interesting behaviour would be to detect if a local data file do already exist and display correctly the hidden password when we click on "show password".

janmulder, 2017-07-26:
Ok, this is one we have to get @dirkhh's attention.

First, yes, the behavior is a little confusing, but @Ghalr's statement that "we have to type the password again" is simply wrong. It is the result of a recent commit  680022aea581. Trying to hide the passwd also removes the astrix characters, so the passwd field seems empty, apparently confusing (some) users.

I propose the following: set the editmode to PasswordEchoOnEdit, allowing the user to see the passwd as it is entered, and remove the switch "show passwd", as we like to hide the passwd from the UI, so there is no need for a switch. I'm happy to create a commit for this. 

Finally, the suggestion "to detect if a local data file do already exist" is void to me. There is never a local data file (or better: git repo) when the app is not installed, unless an (experienced) Android user with root permission made a clone of the repo before removing the app. For this use case, the user is on his own.

Ghalr, 2017-07-26:
This is the part I do not get. If the password is hiddent (astrix characters) what is the point to click on "Show password" if we have the password cleared and we have to type it again. When I do enter Wifi code, the code is hidden and when I click on "Show password" password is revealed without to have it cleared. I just checked to be sure and this is the behavior most users use.

dirkhh, 2017-07-26:
We had a number of concerns that showing an already entered password is a security issue; while I find some of the logic there contrived, I can see why people worry about this - and showing the entered password doesn't really provide value. What is useful is to be able to enter and edit the password without having it hidden by stars - and then, once it is correct, have it no longer shown.

The current implementation isn't correct, yet. But the problem is that we desperately try not to have a 'save changes' button and for this interaction that's just stupid and leads to confusing situations.

I had started to implement what I considered a better work flow but got side tracked because I wanted some of the fields to only be visible when they were needed and that doesn't work with the QML Layout as it doesn't appear to reposition the other items when one of the items inside the layout changes visibility - which defeats the purpose, imho.

janmulder, 2017-07-26:
> What is useful is to be able to enter and edit the password without having it hidden by stars - and then, once it is correct, have it no longer shown.

And that is exactly what editmode: PasswordEchoOnEdit does.

dirkhh, 2017-07-26:
happy to rip out my code. please send PR.
but regardless of this aspect - the current email/password entry is not covering all the use cases that we need and is the thing that prevents me from doing a new beta

closed 2017-07-26 by dirkhh

janmulder, 2017-07-26:
and @dirkhh. Can you give (a list of) the missing use cases related to credentials?

dirkhh, 2017-07-26:
I did this on one of the other issues already. We need a button that actually says "now take this data" instead of the weird "editing finished logic".

For example, if I add an email address and no password, we try to connect to the cloud with that (which of course fails), but we then open the local cache for that account if it exists (we show the notification that we read NN dives from local cache). But then when you close the settings dialog and go to the dive list, we show the credential screen again, because the credentials are invalid.

Anyway, here are the cases that I remember:

First use, no data is there
- pick NO CLOUD
- XOR enter email and password, tap button to create/verify
- IFF new account, show message to expect email with PIN and add PIN field
        -- when PIN is entered and button tapped, validate the PIN, if successful, hide PIN, otherwise repeat
- IFF existing account and correct password, show that credentials were accepted
- otherwise show the error message (or paraphrase it) and keep email but clear password field and ask the user to try again

After the user started using Subsurface-mobile
- NO CLOUD mode, adding email - similar flow to above, but... if we create a new account, store the local dive list in that new account; if we open existing account, ask if this is supposed to replace the data or merge the data
- with existing account, close the existing account, switch to new account (similar to above, deal with creation of new account vs. logging into existing account)

PR #514 (2017-07-26): QML UI: remove "show password" switch and only show pw on first entry
Like the subject says. We do not want the password to be made visible, so a switch to show it, is useless and is therefore removed. Further, the entry mode is set to PasswordEchoOnEdit, which causes the passwd to be visible (for easy entry), but can't be made visible again after save/end edit.

Fixes: #512
commit bb7dcc25 (2017-07-26): QML UI: remove "show password" switch and only show pw on first entry

Like the subject says. We do not want the password to be made
visible, so a switch to show it, is useless and is therefore
removed. Futher, the entry mode is set to PasswordEchoOnEdit,
which causes the passwd to be visible (for easy entry), but
can't be made visible again after save/end edit.

Fixes: #512