subsurface/subsurface #513: [Bug] Mobile: Starting the app display 2 differents background color (Black then White)

Ghalr, 2017-07-26:


janmulder, 2017-07-26:
Also here. Can you show a screenshot of what you exactly mean?

Ghalr, 2017-07-26:
I'll try to put a gif let me see if I can do it easily

Ghalr, 2017-07-26:
Apologize about the video quality my Android is not rooted and bad internet.
https://drive.google.com/file/d/0B6-wHPDkD8URWE5hWThGRXhJTXc/view?usp=sharing

You got 2 background color (Blanck and White) 
and also for the previous bug
https://github.com/Subsurface-divelog/subsurface/issues/512

janmulder, 2017-07-26:
Ok, clear. You are talking about the startup splash screens. Indeed, 2 different background colors. I know this is already for a very long time, this way. Bu yes, this seems a (imho low priority) bug.

Ghalr, 2017-07-26:
Startup splash screens indeed.
Editing the title.....

Ghalr, 2017-07-26:
@janmulder mind if I delete the video link, as you confirmed the issue, I notice there are my personnal email on it... lol

janmulder, 2017-07-26:
Sure, delete the screencast if you like.

neolit123, 2017-09-23:
@janmulder do you think this is an easy fix?

janmulder, 2017-11-17:
New screencast to show the issue. Gunzip first :-)

[black-white.mp4.gz](https://github.com/Subsurface-divelog/subsurface/files/1481496/black-white.mp4.gz)

PR #827 (2017-11-17): mobile: black and white backgrounds on starting
### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

This very simple commit has a long story. See the mailing list
subject "mobile: android splash screen (and issue 513) opacity
weirdness".

There is even more. I found the following deccription on Internet:
https://falsinsoft.blogspot.nl/2017/07/qml-show-android-native-splash-screen.html
and tried to implement this with the pot [... cut]
commit 35194c70 (2017-11-17): mobile: black and white backgrounds on starting

This very simple commit has a long story. See the mailing list
subject "mobile: android splash screen (and issue 513) opacity
weirdness".

There is even more. I found the following deccription on Internet:
https://falsinsoft.blogspot.nl/2017/07/qml-show-android-native-splash-screen.html
and tried to implement this with the potentially nice Qt functionality
(since 5.8) to manually get rid of the splash screen. Unfortunately,
this does not work. Notice that there are subtile differences in
the here found internet page, and the method originally im [... cut]
closed 2017-11-17 by neolit123