subsurface/subsurface #515: [idea] Mobile UI #13: User Auth management

Bocio, 2017-07-26:
Follows comments on #492 (settings page)  I move user authentication management on mobile app here.

From the UI point of view current user authentication is flawed under several aspects. Most evident is the use of one stateless form to manage a stateful process as user authentication.
Trying to keep is simple as possible, some use cases are manually managed by @dirkhh (i.e. password recovery/reset) hence hereafter not all of them are described.
Disclaimer: given that I don't know which messages are exchanged with the cloud server for cloud authentication, I deduced part of them just using the app. Sorry in advance for my bloopers.

Prerequisites:
- The first time the user open the app he is required to register with the Subsurface cloud.
- The user may skips registration and works offline locally. He can access later the user management UI via the settings panel.
- The user may have already a valid cloud account or he wants to register a new account.
- The proposed solution foresee the use of one UI form for both user registration or simple sign-in and one optional UI form for PIN code.

### Use state
AFAIK there are three possible states both on server and app.
1. **no user defined**: first time install or the user uninstalled and installed again the app 
2. **user unconfirmed**: the user correctly registered on the served but he has not been validated through the PIN code.
3. **user registered**.
**N.B.**
We may have user state 1 on app and user state 3 on server.

### Activity diagram for the proposed user auth management
- blue state/actions require UI
- yellow state/actions require a message to the user
- for the sake of simplicity wrong password on server is not depicted.

[image]

### User & pwd UI first time install or registration

This panel can be accessed on first time app launch or via settings cloud account management. There is no user account stored on app hence the UI ask to enter a valid email and password.

[image]

The app validates the passwordfor forbidden chars and against the server. Eventually an error message is given to the user.
The app registers for user account on the cloud. 
1. No user is found on the server for the given email so a new user is created and a PIN code is sent. (see the PIN code UI)
2. A valid user/password is found on the server. This means that the user already registered and installed the app again or he has multiple account and he is switching account.

### PIN code UI

This panel is accessed when the user is in state **unconfirmed**. Hence:
1. It's step 2 after a new user registration has been performed on the server.
2. He accesses to the cloud account management via the Settings panel and he has never entered PIN code.

[image]

### User & pwd UI already registered

User can access to the sign-in or registration panel even if he is already registered with the cloud. He can switch user or he want to create a new user.
UI panel it's the same. We only give the user a different help text.

[image]

dirkhh, 2017-08-01:
The flow chart at the top is missing "wrong password" for the "[registered user]" case
I think we got rid of the show password option and instead use the standard "show as you enter" thing.
I'm ok with the PIN being its own screen - that should be a very rare thing for people to interact with.

A few flows are missing:

- user choses NoCloud and then later wants to connect to an account. I guess from a UI perspective this might be the same flow, but we need to make sure that there are no gotchas
- user wants to switch to a different account
- user wants to change their password

Bocio, 2017-08-02:
> The flow chart at the top is missing "wrong password" for the "[registered user]" case

Yes I specified that _for the sake of simplicity wrong password on server is not depicted._ BTW I think that a wrong password will implies just a toast to the user.

> A few flows are missing:
> 
> user choses NoCloud and then later wants to connect to an account. I guess from a UI perspective this might be the same flow, but we need to make sure that there are no gotchas.

I forecast this.
_## User & pwd UI first time install or registration
This panel can be accessed on first time app launch or via settings cloud account management. There is no user account stored on app hence the UI ask to enter a valid email and password._
_The user may skips registration and works offline locally. He can access later the user management UI via the settings panel._

Basically if the user skips registration he may reach the user management UI later via the settings panel. The UI it's always the same.

> user wants to switch to a different account

Yes I expected it already. It's the _"User & pwd UI already registered"_ paragraph. Basically it's the same UI with different hint text.

> user wants to change their password

I forgot to mention this. Actually I remember that it's one of that functionalities you manage manually. If you want to automate this I guess we have to provide an explicit action/UI.

PS
We have to get this part right. Since news of the new beta download features spreaded over the internet I saw several users asking help on authentication and they will increase at the official launch.

dirkhh, 2017-08-02:
> 
> user wants to change their password
> 
> I forgot to mention this. Actually I remember that it's one of that functionalities you manage manually. If you want to automate this I guess we have to provide an explicit action/UI.
> 
BTW: password CHANGE on the desktop app works. What isn't automated is password RECOVERY.
i.e., if you know your old password, you can change to a new one - that is supported. But if you forgot your password, you can't set a new one without my intervention.
I need to implement something for this on the backend.

/D

Bocio, 2017-09-26:
@dirkhh @neolit123 IMHO MHO IMHO this is the only real issue we should implement before a new main official release.

janmulder, 2017-09-26:
@dirkhh, @Bocio, @neolit123: working on it as we speak ... and it is non trivial ... so no promises on date ready.

dirkhh, 2017-09-26:
@janmulder 👍 👍 👍 👍 👍

dirkhh, 2017-09-26:
And btw: since I am stuck with BLE on iOS right now, I have no idea of timeline. I'd love to release 2.0 on both platforms. In some ways, I might even release 2.0 on iOS with no BLE download - just because the UI is so much better. But I definitely want to try a little more to figure out what's going wrong there.

Bocio, 2017-10-16:
Thanks to the amazing work of @janmulder on version 2.0 I guess we could close this. Regardless of bugs, as idea it's fully implemented. Do you agree?

janmulder, 2017-10-16:
I agree @Bocio. Indeed, despite the fact that there are still some lose ends and bugs, but I feel that it is better to close this (great) idea now, and address remaining bugs and lose ends in separate issues. So closing it.

closed 2017-10-16 by janmulder