subsurface/subsurface #521: 4.6.4: CTRL-V while editing dives not working

madscientist, 2017-07-29:
Subsurface 4.6.4 / Linux

I'm adding dives from a trip I took last year, by hand (no computer).  I added the first dive then used CTRL-C (Copy dive components) to make a copy of things like cylinders, weights, buddy, and suit.

Then I use CTRL-+ to create a new dive and enter date, time, depth, duration, location

Then I try to use CTRL-V to paste the copied components.  Nothing happens.  I've made sure that I'm not in the middle of entering another value (e.g., there is no drop-down box active), but nothing happens.

If I use the menu option (Log -> Paste dive components) it works fine.

Also if I first click "Apply Changes", then use CTRL-V, that works as well (the dive is edited, components are updated and I can click "Apply Changes" again to confirm).

madscientist, 2017-07-29:
I was just able to get CTRL-V to work once, but then when I tried to reproduce it I couldn't.  There's definitely something odd about the keybindings.

kapitainsky, 2017-08-07:
very similar on mac OS 10.12.6 with Subsurface 4.6.4. I have not figured out yet how to reproduce it but it always works using the Log menu. Interestingly when shortcuts for copy/paste don't work, in Log menu cmd-C and cmd-V shortcuts indicators are missing too. Restarting Subsurface solves the problem.

neolit123, 2017-09-23:
@tcanabrava hi, Tomaz. do you know why this can happen?

bstoeger, 2020-05-03:
I'm closing this, since we don't even have a separate "add dive" mode anymore. If there is a general copy & paste problem, please open a new bug report.

closed 2020-05-03 by bstoeger