subsurface/subsurface #534: [idea] Add Divesoft CCR Liberty GPS Integration

zamedic, 2017-08-07:
Subsurface already imports the Divesoft Liberty and Freedom DLF files. Recently, Divesoft released a GPS Module for the Liberty CCR. 

It would be cool if the GPS Co-ordinates for the dive can be automatically imported when the DLF file is imported.

glance-, 2017-08-07:
The DLF import haven't seen any love in a while, and a quick glance at my old notes doesn't say anything about where a GPS coordinate could be stored so if you have some example files or some information about where in the file that can be found it would be a good start.

zamedic, 2017-08-07:
Yeah - I see so as well. 

Here are some files. 
[dives.zip](https://github.com/Subsurface-divelog/subsurface/files/1205321/dives.zip)

81 - Lat: 24.938144°, Lon: 35.866066°
80 - Lat: 24.938853°, Lon: 35.863803°
79 - Lat: 24.931987°, Lon: 35.867736°

glance-, 2017-08-07:
NDL, TTS looks bonkus and Temperature looks a bit wierd to. Those are right next to each other in the sample.

zamedic, 2017-08-07:
I suspect that somewhere along the line, divesoft must have updated their file format. When I upload to wetnotes.com (divesoft's site) - it appears to accept the data correctly. Other "commercial" softwares also doesn't interpret the data. Want me to reach out to divesoft and see if I can get the new file layout from them?

zamedic, 2017-08-07:
I compared the Divesoft Freedom and the Divesoft Liberty files (I have both)... Subsurface is accepting the data from the Freedom and interpreting it correctly, the Liberty the fields are a bit messed... So I will need to create a separate issue to have submerge determine if the file is Freedom or Liberty.

glance-, 2017-08-07:
I found the gps data atleast, and a bit which seems to indicate that ndl/tts/temp are woknus.

I did some other dlf stuff to, all in a coming PR.

zamedic, 2017-08-08:
Wow - you rock :) thank you

👍

closed 2017-08-10 by dirkhh

zamedic, 2017-08-15:
@glance- I have done some testing with the Liberty and Freedom - looks like it working perfectly. Thank you.

glance-, 2017-08-15:
I wrote a email to Divesoft, but the mail adress to our old contact there bounced, so I sent it to info@divesoft.cz to. No answer yet.

zamedic, 2017-08-15:
I spoke to them last week - their technical department is on leave and will only be back next month... 

I have also sent them a link to the beta to ask them to test with some data they have. 

> our team for technical support are on holiday.
> They will contact you on September.
> Thank you for your understanding.

PR #535 (2017-08-10): Improve DLF import
This does some minor work to the DLF importer, as a result of the bug report and test data we got in #534.

This should fix #534.