subsurface/subsurface #538: 4.6.4 and Android v2: Insufficient privileges to open the device Shearwater (Perdix), via BT

hildegw, 2017-08-09:
Both via Android v2 and Subsurface 4.6.4 on Ubuntu I get the above message. The Perdix pairs on BT, and shows up as device on Subsurface, but no data is being read/transmitted. 

on Ubuntu I get the following error (MAC address changed):
"Insufficient privileges to open the device XX:56:2C:C8:XX:XX Shearwater (Perdix) "
and console output:
"qt.bluetooth.bluez: void QBluetoothSocketPrivate::_q_readNotify() 23 error: -1 "Resource temporarily unavailable"
Connection on channel 1 failed. Trying on channel number 5.
qt.bluetooth.bluez: void QBluetoothSocketPrivate::_q_readNotify() 23 error: -1 "Resource temporarily unavailable"
Failed to connect to device  ED:56:2C:C8:FC:29 . Device state  QBluetoothSocket::UnconnectedState . Error:  QBluetoothSocket::UnknownSocketError
[10.270901] ERROR: Failed to open the serial port. [in shearwater_common.c:46 (shearwater_common_open)]"

Pretty much the same on Android device, here are the logs.
[subsurface.log.txt](https://github.com/Subsurface-divelog/subsurface/files/1212958/subsurface.log.txt)
[libdivecomputer.log.txt](https://github.com/Subsurface-divelog/subsurface/files/1212959/libdivecomputer.log.txt)

I can read out the Perdix with my Android device via BT from Shearwater Cloud, so BT connection works.

data2, 2017-08-14:
I have the same issue after upgrading my Petrel 2 to v44 from v37. I assume this has to do with the change in v40: https://www.shearwater.com/wp-content/uploads/2017/07/Release-Notes-Translations-FINAL.pdf

"Only one way to start Bluetooth now. Removed the “Upload
Log” and “Load Upgrade” menus. The new “Start Bluetooth”
option is at the top-level of menus."

neolit123, 2017-09-23:
@glance- @dirkhh

dirkhh, 2017-09-23:
We seem to have problems with dual stack (BT & BLE) Petrel 2 dive computers to do BT based connections. I don't think this has anything to do with the way BT is now started - we do something wrong when trying to connect to it.
And strangely it works on Windows (which uses somewhat different BT code in Subsurface) but seems to fail on Android, Linux and Mac

data2, 2017-09-24:
After having conferred with the Shearwater devs, they recommended the obvious: deleting and re-pairing the dive computer. That worked for me.

dirkhh, 2017-09-24:
Oh good. That's not fixing it for me right now on the Mac, but we can close this issue then as apparently my problem is different.

Thanks for the fast response!

closed 2017-09-24 by dirkhh