subsurface/subsurface #542: AppImage comes without SELinux and with Qt4

ShalokShalom, 2017-08-15:
Hi there ^-^

I recently tried to download and install... ääh run your AppImage and get an error regarding SELinux, which is not installed here. 

Our distro gets maintained by one single person in order to focus on Qt/KDE, instead of putting it on top of a GTK+ focused stack and in order to maintain that that is the repository focused on quality instead of quantity, so it comes that we use the hardened Linux as security bonus. 

There is no SELinux available and while most distros seem to ship it, is the approach of AppImage to work on all distros - to make SELinux now as a requirement, is a bit strange, IMHO.

Especially since it's developed by the NSA, while guess this is simply paranoid from my side. 

The next thing is, that Qt4 seems to be either part of the AppImage or required by the app from the system as well: My question here is: Why? 

Qt5 gets released in 2012, Qt4 is unmaintained since 3 years or so and we dont ship it since one year. 
I guess every single distro ships Qt5 these days and i think Subsurface itself is also already fully ported, correct?

dirkhh, 2017-08-15:
The AppImage / SELinux dependency is really annoying - but I haven't figured out how to remove that, yet.
There should be NO Qt4 dependency. Subsurface doesn't even compile against Qt4 anymore. Are you sure this is what you are seeing? If yes, can you provide a bit more information so I can try and track that down?

ShalokShalom, 2017-08-15:
I see 32Bit libs also, while I downloaded the 64 Bit version:

```
➜  Downloads git:(master) ✗ /home/matthias/Downloads/Subsurface-4.6.2-x86_64.AppImage 
Started in /home/matthias/Downloads ...
Moving to /tmp/.mount_1moIx9 ...
Executing '/tmp/.mount_1moIx9/usr/bin/subsurface.wrapper' ...
counting args in %1$s/usr/lib/:%1$s/usr/lib/i386-linux-gnu/:%1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/:%1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib64/:%2$s
  using LD_LIBRARY_PATH=/tmp/.mount_1moIx9/usr/lib/:/tmp/.mount_1moIx9/usr/lib/i386-linux-gnu/:/tmp/.mount_1moIx9/usr/lib/x86_64-linux-gnu/:/tmp/.mount_1moIx9/usr/lib32/:/tmp/.mount_1moIx9/usr/lib64/:/tmp/.mount_1moIx9/lib/:/tmp/.mount_1moIx9/lib/i386-linux-gnu/:/tmp/.mount_1moIx9/lib/x86_64-linux-gnu/:/tmp/.mount_1moIx9/lib32/:/tmp/.mount_1moIx9/lib64/:
  using PATH=./bin/:./sbin/:./games/:../bin/:../sbin/:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
  using PYTHONPATH=./share/pyshared/:
  using XDG_DATA_DIRS=./share/:/home/matthias/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share
counting args in %1$s/usr/lib/qt4/plugins/:%1$s/usr/lib/i386-linux-gnu/qt4/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/qt4/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib64/qt5/plugins/:%2$s
  using QT_PLUGIN_PATH=/tmp/.mount_1moIx9/usr/lib/qt4/plugins/:/tmp/.mount_1moIx9/usr/lib/i386-linux-gnu/qt4/plugins/:/tmp/.mount_1moIx9/usr/lib/x86_64-linux-gnu/qt4/plugins/:/tmp/.mount_1moIx9/usr/lib32/qt4/plugins/:/tmp/.mount_1moIx9/usr/lib64/qt4/plugins/:/tmp/.mount_1moIx9/usr/lib/qt5/plugins/:/tmp/.mount_1moIx9/usr/lib/i386-linux-gnu/qt5/plugins/:/tmp/.mount_1moIx9/usr/lib/x86_64-linux-gnu/qt5/plugins/:/tmp/.mount_1moIx9/usr/lib32/qt5/plugins/:/tmp/.mount_1moIx9/usr/lib64/qt5/plugins/:
  using PERLLIB=./share/perl5/:./lib/perl5/:
  using GSETTINGS_SCHEMA_DIR=./share/glib-2.0/schemas/:
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Available platform plugins are: linuxfb, minimal, offscreen, xcb, wayland-org.kde.kwin.qpa, eglfs, minimalegl, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx.

Reinstalling the application may fix this problem.
/tmp/.mount_1moIx9/usr/bin/subsurface.wrapper: Zeile 121:   355 Abgebrochen             (Speicherabzug geschrieben) LD_LIBRARY_PATH="" kdialog --caption "" --title "$TITLE" -yesno "$TEXT"
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

[... cut ...] vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx.

Reinstalling the application may fix this problem.
/tmp/.mount_1moIx9/usr/bin/subsurface.wrapper: Zeile 121:   359 Abgebrochen             (Speicherabzug geschrieben) LD_LIBRARY_PATH="" kdialog --caption "" --title "$TITLE" -yesno "$TEXT"
/tmp/.mount_1moIx9/usr/bin/subsurface: error while loading shared libraries: libselinux.so.1: cannot open shared object file: No such file or directory
➜  Downloads git:(master) ✗ 

```

ShalokShalom, 2017-08-15:
> The AppImage / SELinux dependency is really annoying - but I haven't figured out how to remove that, yet.

The AppImage developer writes, that this is because you bundle Subsurface on a distro that uses SELinux.

dirkhh, 2017-08-15:
I wonder what a good base distro would be that doesn't have SELinux. We currently use CentOS 6. I'm open to other ideas (but it would have to be something reasonably old in order for the AppImage to work on older distros)

⁣-- 
From my phone​

https://github.com/Subsurface-divelog/subsurface/issues/542#issuecomment-322393882

ShalokShalom, 2017-08-15:
I guess an ISO which is some years old does not count as old and valid?

I guess Slackware is then maybe something to consider. 
While this is still too young in order to provide support for CentOS 6. 

What is about simply disabling SELinux? https://kerneltalks.com/linux/disable-selinux-linux-distro/

ShalokShalom, 2017-08-16:
@dirkhh  https://github.com/AppImage/AppImages/issues/83#issuecomment-322452418

ShalokShalom, 2017-09-08:
Its enough to include libselinux.so.1

ShalokShalom, 2018-05-10:
Is this solved now?

mikeller, 2026-10-17:
Hi @ShalokShalom,

This report is tied to the CentOS6-based AppImage build image from 2017, the AppImage build base has moved on substantially since then, and SELinux/Qt-bundling practices in AppImages generally have also evolved. With no further detail and no corroborating reports in years, we're closing this as stale/superseded.

Please reopen with your current AppImage version and distro if you're still seeing missing SELinux support or an unexpectedly old bundled Qt.

closed 2026-10-17 by mikeller