subsurface/subsurface #547: [idea] mobile - multiple gasses

rainman104, 2017-08-17:
Probably a low priority item and not the end of the world and I realize it would possibly clutter up the UI, but would love to be able to save multiple tanks.   Currently it appears I can only define a single tank for a dive I'm entering in on mobile.

neolit123, 2017-09-23:
@janmulder hi jan, do you think this change-request is feasible?

janmulder, 2017-09-23:
@neolit123: the topic starter made some very valid remarks. Low prio, possible clutter of the UI. I believe that we need to be careful with respect to function creep. There will never be equal functionality compared to the desktop app. So, we will always have users that like some functionality that is currently desktop only also on the mobile version. 

This all said. Yes, this idea might be possible but screen real estate is (very) limited, and there are much more relevant tasks at hand to get mobile version 2.0 live, so I would propose: close, or very low prio, and to be solved in a more complete overhaul of manual dive logging on mobile.

neolit123, 2017-09-23:
ok, i suggest we leave it for now and see in the following months how the mobile ui will develop.
i've assigned the issue to you, so you can close it if you'd like.
thanks.