subsurface/subsurface #548: Importing from Shearwater Desktop does not preserve tank pressure data

SeppoTakalo, 2017-08-20:
I'm using Shearwater Perdix AI with tank pressure transmitter installed.
Pressure data is downloaded into Shearwater Desktop application and visible there.
I cannot find a method to import this data into Subsurface.

## Import method
1. On Shearwater Desktop go to File->Backup database.
1. On Subsurface: Import->Import log files. Select the previously saved `ShearwaterDesktop.db` file.

## Result
All other data seems to be imported, except tank pressures.
Also, the log view shows the computer as a "Shearwater Petrel" even that it is Perdix AI. I'm not sure if this is relevant.

mturkia, 2017-08-20:
What version of Subsurface you are using? I have added support for 1 pressure sensor in May, but that is not yet in official release. If you send me a sample log, I could also take a look into the DC name.

SeppoTakalo, 2017-08-20:
Subsurface version 4.6.4

Here is the log file that I was trying to import. Please note that not all dives had the AI installed and therefore the data is missing on most of those.
[ShearwaterDesktop.db.zip](https://github.com/Subsurface-divelog/subsurface/files/1237344/ShearwaterDesktop.db.zip)

mturkia, 2017-08-22:
Can you give me also information about the correct pressures? (It seems that the database has same DC model for both Petrel and Perdix, at least if my

SeppoTakalo, 2017-08-22:
Not all dives have the pressure information but here are the ones that have:

|dive|start pressure (bar)|end pressure (bar)|
|----|--------------------|------------------|
|177|71.43|62.6|
|176|119.14|72.12|
|169|202.02|0.98|
|168|190.85|130.31|

Also, here is the XML exports from those dives. They seem to have PSI pressures inside the log files per sample.
[exports.zip](https://github.com/Subsurface-divelog/subsurface/files/1241199/exports.zip)

commit a20aa850 (2017-08-24): Support for air pressure on Shearwater Desktop import

Fixes #548
closed 2017-08-24 by dirkhh