subsurface/subsurface #550: Stichwörter werden nicht übernommen und aktiv gelöscht beim Feldwechsel

Natas-Lived, 2017-08-24:
Ein neuer Tauchgang wird manuell angelegt: STRG++ drücken und alle Daten eingeben. Im Feld "Stichwörter" werden ebenfalls Daten eingegeben. Mit Tab in das Notizenfeld springen und Tauchgang speichern (mit der Maus, "Änderungen anwenden" Knopf).
Jetzt sieht man, daß die Daten im Feld "Stichwörter" verschwunden sind.
Gleiches gilt auch, wenn man statt zu Speichern ("Änderungen anwenden") auf die Tabulatorseite "Ausrüstung" wechselt und dort Daten eingibt. Klickt man wieder auf die Seite "Notizen", ist das zuvor gefüllte Feld "Stichwörter" wieder leer.
Trägt man (in beiden Varianten) nun abermals Daten in das Feld "Stichwörter" ein und speichert, bleiben die Daten erhalten und werden in das Tauch-Log eingetragen.

Erwartet
Trägt man Daten in ein Feld ein, so sind diese nicht zu löschen, wenn man in andere Felder springt oder die Tab-Seite wechselt.
Eine doppelte Eintragung sollte nicht notwendig sein.

dirkhh, 2017-09-23:
also @tcanabrava

Natas-Lived, 2017-09-26:
english translation provided.

dirkhh, 2017-09-26:
Thanks @Natas-Lived 
@tcanabrava can you comment if this is something you could look into?

bstoeger, 2017-12-20:
I tried and tried, but was unable to reproduce this (current master, Kubuntu, Qt 5.9.1).

Could you perhaps provide a screen capture in case that I'm not understanding the description correctly?

Natas-Lived, 2017-12-29:
Attached a png with the forms prior and after pressing "save" i.e. with data and and with lost data. 

[image]

mturkia, 2017-12-30:
What version of Subsurface? Is it still an issue with 4.7.6 that was released yesterday?

bstoeger, 2020-05-03:
Since we don't even have the "save changes" dialog anymore, we probably should close this.

dirkhh, 2020-05-03:
agreed

closed 2020-05-03 by dirkhh