subsurface/subsurface #551: Filter wird gelöscht bei Anpassung der Werte im Filterfeld

Natas-Lived, 2017-08-24:
Die Filteransicht ist aufgerufen (STRG+F)
Der Filter ist auf "Anzug".
Es wird nach dem Eintrag "Nass" gefiltert, einige Daten werden angezeigt.
Der erste Datensatz wird bearbeitet und Naß eingetragen. Klick auf "Änderungen anwenden".
Der Filter wird entfernt und es werden wieder alle Daten (Tauchgänge) angezeigt.

erwartet:
Liste der Tauchgänge mit "Nass" wird weiterhin angezeigt, jedoch mit einem Datensatz weniger.
in der Übersicht der möglichen Filterkriterien wird neben Nass jetzt auch Naß angezeigt, mit einem Datensatz als Anzahl.

dirkhh, 2017-09-23:
This is something that  @tcanabrava would be the best person to work on.
Tomaz, do you understand the German? Want translation help?

dirkhh, 2017-10-02:
Quick translation:
Open the filter view (Ctrl-F)
Set the filter for a "suit"
Filter for the first entry under suit (in the German example "Nass" - using sample data, try "wet, 3mm")
Edit the first dive that is selected and shown, change the suit (let's say to "wet, 2mm") and accept the change
Result: filter is removed and all dives are shown
Expected result: still only those dives with suit "wet, 3mm" are shown, just one fewer dive, and the list of options in the filter panel should now also offer "wet, 2mm"

bstoeger, 2017-11-25:
Fixed by #857 and #859? Please check.

commit 6343515f (2017-11-25): Use equality instead of substring comparison in suits and buddy filter

This commit is a continuation of commit 739b27427cfb5119eebe214c984843cd5d155620,
in which a substring comparison was replaced by equality comparison to
avoid confusing UI behavior of the filter interface.

The suit and buddy filters were plagued by the same problem, so change
their code in analogy.

Fixes #551 (in conjunction with commit dd2466f51899aae406dc8c13904787710f30ec1c).
closed 2017-11-25 by dirkhh