subsurface/subsurface #554: falsche Daten bei manuellen Eintragungen

Natas-Lived, 2017-08-31:
Ein neuer Tauchgang wird manuell eingetragen, die Dauer kann man ebenfalls angeben.

Nach Speichern wird in der Liste der Tauchgänge ein Wert um einen Zähler höher angezeigt. Aus 37 Minuten in der Eingabemaske wird 38 Minuten in der Tabelle.

dirkhh, 2017-09-01:
Welche Subsurface Version? Welches Betriebssystem?

Natas-Lived, 2017-09-04:
Version 4.6.4, Linux, Suse 42.1.20170515
Bildschirmabbild anbei
[image]

Natas-Lived, 2017-09-04:
Version 4.6.4
BS: Linux Suse 42.1
Bildschimabbild anbei
[image]

dirkhh, 2017-09-04:
Hmm, ok, das sieht aus, als wuerden wir unterschiedlich runden. Kannst Du den einen Tauchgang, den Du da zeigst, als XML exportieren und hier anhaengen, damit wir das einfach reproduzieren koennen?

Natas-Lived, 2017-09-07:
Datei angehängt, Endung .txt löschen, ohne unterband GitHub das Hochladen.
[554_falsche_Daten_bei_manuellen_Eintragungen_Export_TG27.ssrf.txt](https://github.com/Subsurface-divelog/subsurface/files/1284491/554_falsche_Daten_bei_manuellen_Eintragungen_Export_TG27.ssrf.txt)
Es ist ein manuell eingetragener Tauchgang:
<dive number='27' (...) duration='37**:06** min'>
Woher die sechs Sekunden kommen, kann ich nicht erklären, eingegeben hab ich sie nicht.
Die Werte in Sekunden sind auch bei anderen manuell eingetragenen Datensätzen enthalten.

bstoeger, 2018-06-15:
I think this bug is fixed:

1) After ac1ec486ae0cd0295166bb1dc0d5b82dc237a5b1, no 6 second are added to the dive duration
2) After eb38a641490752578e3dddc7b7959cfd52e0be74, the dive duration is not rounded if there are seconds.

The duration in the dive-list also seems to work: X:YY is rounded to X for Y < 30 and to X+1 for Y >= 30.

Unfortunately, I don't see how we can fix those erroneous +6s dives?

neolit123, 2018-06-15:
> I think this bug is fixed:

ok, lets close this and open new issues if needed.

closed 2018-06-15 by neolit123