subsurface/subsurface #560: Nextcloud, Dropbox or other Clouds

bes1002t, 2017-09-06:
Is it possible to use own cloud solutions like nextcloud or dropbox with subsurface?

atdotde, 2017-09-06:
When using the desktop version of Subsurface, you can have your log in an xml file and nothing speaks against having that in a Dropbox shared folder. The only possibility, though, to share data with the mobile version is via the Subsurface cloud.

neolit123, 2017-10-11:
closing this issue due to inactivity.

closed 2017-10-11 by neolit123

mrpew, 2022-07-28:
> When using the desktop version of Subsurface, you can have your log in an xml file and nothing speaks against having that in a Dropbox shared folder. The only possibility, though, to share data with the mobile version is via the Subsurface cloud.

Does the mobile version store the divelog differently? Unfortunately I have 0 experience with Android apps developed in QT, are there API limitations that prevent/increase the complexity of a "divelog xml file location" setting in the mobile app, similar to the desktop version?

It seems like a setting for the XML file location would allow compatibility with most cloud options, without the need for specific implementations for individual providers. 

Id generally be open to look into this and contribute, as I prefer

dirkhh, 2022-07-28:
Having multiple apps simultaneous be able to have an XML file open and write to it is a terrible idea.
The whole point of our cloud solution is to use git as backend in order to enable distributed use of the date.
This would create orders of magnitude more problems than it would solve. Please don't do that.

mrpew, 2022-07-28:
That makes a lot of sense, thank you for clarifying!