subsurface/subsurface #561: all kinds on upgrading logbook from version 2 to 3

antiker1, 2017-09-08:
Hi.

I was just trying to upgrade my logbook from version 2 to that of current Subsurface 4.6 (i.e. version 3).

That seems to imply all kinds of corruptions and data losses in the XML, and even when I manually diff the old/new one and get manually merge the correct information into the version 3 XML, after another saving in 4.6, subsurface seems trying to be smart (but fails miserably) and does the same corruptions again... o.O

Some examples, without the stuff that is changed (or possibly broken) related to gaschanges:
```
$ diff manually-corrected.ssrf after-saving-again.ssrf | grep -v gaschange
85,86c85,86
< <trip date='2012-06-06' location='Curaçao; Netherlands'>
< <dive number='1' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-06'>
---
> <trip date='2012-06-06' time='00:00:00' location='Curaçao; Netherlands'>
> <dive number='1' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-06' time='00:00:00' duration='0:00 min'>
95c95
< <dive number='2' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-06' duration='44:00 min'>
---
> <dive number='2' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-06' time='00:00:00' duration='44:00 min'>
116c116
< <dive number='3' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-07'>
---
> <dive number='4' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-07' time='00:00:00' duration='0:00 min'>
119,128c119
<   <notes>PADI Open Water Diver - Confined Water Dive #2</notes>
<   <cylinder size='12.0 l' workpressure='200.0 bar' description='12L 200 bar' />
<   <weightsystem weight='5.4 kg' description='integrated' />
<   <divecomputer model='none'>
<   </divecomputer>
< </dive>
< <dive number='4' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-07'>
<   <divemaster>snip snap</divemaster>
<   <buddy>snip snap</buddy>
<   <notes>PADI Open Water Diver - Confined Water Dive #3
---
>   <notes>(PADI Open Water Diver - Confined Water Dive #2) or (PADI Open Water Diver - Confined Water Dive #3
131c122
< 200 m mask/snorkel/fin swim</notes>
---
> 200 m mask/snorkel/fin swim)</notes>
137c128
< <dive number='5' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-07' duration='40:00 min'>
---
> <dive number='5' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-07' time='00:00:00' duration='40:00 min'>
167c158
< <dive number='6' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-08'>
---
> <dive number='7' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-08' time='00:00:00' duration='0:00 min'>
170c161
<   <notes>PADI Open Water Diver - Confined Water Dive #4
---
>   <notes>(PADI Open Water Diver - Confined Water Dive #4
172c163
< Skin Diving Skills</notes>
---
> Skin Diving Skills) or (PADI Open Water Diver - Confined Water Dive #5)</notes>
178,187c169
< <dive number='7' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-08'>
<   <divemaster>snip snap</divemaster>
<   <buddy>snip snap</buddy>
<   <notes>PADI Open Water Diver - Confined Water Dive #5</notes>
<   <cylinder size='12.0 l' workpressure='200.0 bar' description='12L 200 bar' />
<   <weightsystem weight='5.4 kg' description='integrated' />
<   <divecomputer model='none'>
<   </divecomputer>
< </dive>
< <dive number='8' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-08' duration='36:00 min'>
---
> <dive number='8' tags='salt, shore, student' divesiteid='7d9e9820' date='2012-06-08' time='00:00:00' duration='36:00 min'>
214c196
[... 395 lines cut ...]
>   <sample time='51:20 min' depth='0.02 m' />
>   <sample time='51:24 min' depth='0.02 m' />
>   <sample time='51:28 min' depth='0.02 m' />
>   <sample time='51:32 min' depth='0.02 m' />
>   <sample time='51:36 min' depth='0.02 m' />
>   <sample time='51:40 min' depth='0.02 m' />
>   <sample time='51:44 min' depth='0.02 m' />
>   <sample time='51:48 min' depth='0.02 m' />
>   <sample time='51:52 min' depth='0.02 m' />
>   <sample time='51:56 min' depth='0.02 m' />
>   <sample time='52:00 min' depth='0.02 m' />
>   <sample time='52:04 min' depth='0.02 m' />

```

As you can see from the above diff (I consider the logboo private and would rather not share it including all its details) the following is messed up:
- Wrong times and durations are automagically added (and set to 00:00:00 / 00:00) for which no times/durations are known - however, none of these dives took place at midnight and none was only 0s long.

- Dives 3 and 6 are merged into 4 and 7 respectively, thereby making both corrupted.

- It completely messes up and depth records with negative numbers (i.e. when the computer was on the surface)... I think it would even be wrong to set such records to 0.0m (because if the computer delivers negative data then is what it delivered and this is what should be stored)... but 0.02m or as it's done further above: pretty random close to 0 numbers is just strange.

Would be nice if these things could be fixed :-)

Cheers!

antiker1, 2017-09-08:
Then issues with respect to cylinder/gaschange records…

1) I have very old dives log entries, for which I have no data at all about the cylinders (not even which size).
Consequently, no cylinder was added.
In the old version 2 XML these log entries still had a:
`   <event time='0:02 min' type='25' value='21' name='gaschange' />`
entry, which is now automagically removed.
This is IMO data loss. cause I *do* know that I breathed air, which is however no lost in the logfile.

2) If often dive with side-mounted tanks (e.g. another 10l air or so) or stage tanks with the same gases, for which I know the tanks, start and end pressures.
Which gives me e.g.:
```
  <cylinder size='12.0 l' workpressure='200.0 bar' description='12L 200 bar' start='200.0 bar' end='90.0 bar' />
  <cylinder size='12.0 l' workpressure='200.0 bar' description='12L 200 bar' start='200.0 bar' end='110.0 bar' />
  <event time='0:02 min' type='25' value='21' name='gaschange' />
```
in the old format 2.

In the new one, subsurface seems to try magic again and automatically adds:
`  <event time='0:02 min' type='25' value='21' name='gaschange' cylinder='0' />`
However, in most cases this is simply wrong, and not it also completely omits that I've actually used two tanks as only cylinder 0 is now set.

Now I read the FAQ entry which suggests:
> How can I use more than one tank with the same gas?
> This is a typical question for side mount divers or some tec divers. Subsurface supports having more than one tank with the same gas, even if some dive computers don’t. Simply add a gas change to your second tank of the same gas as both tanks will be included in the gas use calculations. In order to add gas changes simply right-click on the profile at the appropriate spot and you will be offered to add such an event.

But one cannot do this properly unless one would have actually recorded the change times, which most people never do if its the same gas, as it makes little sense to record this then.
E.g. I switch typically every 50 bars when I have multiple tanks with the same gases. but this is only a rough estimate.
IMO, it would be much better, if subsurface sees that there are multiple tanks with the same gas, and the start\end pressure differs (i.e. the tanks were actually used), and no separate gaschange events have been explicitly made by the user/computer, it does something like:
`  <event time='0:02 min' type='25' value='21' name='gaschange' cylinder='0,1,4' />`
To denote that cylinders 0, 1 and 4 have been used with no further explicit gas change events.
This seems to be the least making up artificial data which like not true at all.

Thanks!

antiker1, 2017-09-08:
Then I've noted the following:
For some log entries, the upgrade from version 2 to 3 adds just the o2 and he values:
v2:
```
  <event time='0:02 min' type='25' value='2555920' name='gaschange' />
  <event time='5:30 min' type='25' value='4128777' name='gaschange' />
  <event time='7:10 min' type='20' name='PO2' />
  <event time='12:44 min' type='25' value='2555920' name='gaschange' />
  <event time='16:30 min' type='25' value='4128777' name='gaschange' />
  <event time='16:56 min' type='25' value='2621457' name='gaschange' />
  <event time='22:49 min' type='25' value='2424868' name='gaschange' />
  <event time='46:52 min' type='11' value='79' name='gaschange' />
  <event time='69:00 min' type='25' value='4128777' name='gaschange' />
  <event time='74:00 min' type='11' value='79' name='gaschange' />
  <event time='89:00 min' type='25' value='4128777' name='gaschange' />
  <event time='94:00 min' type='11' value='79' name='gaschange' />
```

v3:
```
  <event time='0:02 min' type='25' value='2555920' name='gaschange' cylinder='0' o2='16.0%' he='39.0%' />
  <event time='5:30 min' type='25' value='4128777' name='gaschange' cylinder='1' o2='9.0%' he='63.0%' />
  <event time='7:10 min' type='20' name='PO2' />
  <event time='12:44 min' type='25' value='2555920' name='gaschange' cylinder='0' o2='16.0%' he='39.0%' />
  <event time='16:30 min' type='25' value='4128777' name='gaschange' cylinder='1' o2='9.0%' he='63.0%' />
  <event time='16:56 min' type='25' value='2621457' name='gaschange' cylinder='4' o2='17.0%' he='40.0%' />
  <event time='22:49 min' type='25' value='2424868' name='gaschange' cylinder='2' o2='36.0%' he='37.0%' />
  <event time='46:52 min' type='11' value='79' name='gaschange' cylinder='3' o2='79.0%' />
  <event time='69:00 min' type='25' value='4128777' name='gaschange' cylinder='1' o2='9.0%' he='63.0%' />
  <event time='74:00 min' type='11' value='79' name='gaschange' cylinder='3' o2='79.0%' />
  <event time='89:00 min' type='25' value='4128777' name='gaschange' cylinder='1' o2='9.0%' he='63.0%' />
  <event time='94:00 min' type='11' value='79' name='gaschange' cylinder='3' o2='79.0%' />

```

Especially the value attribute stays the same.

On other entries however, the value attribute is changed for some of the gaschange events:
v2:
```
  <event time='0:02 min' type='25' value='2621455' name='gaschange' />
  <event time='0:02 min' type='20' name='PO2' />
  <event time='21:56 min' type='20' name='PO2' />
  <event time='21:56 min' type='25' value='2555937' name='gaschange' cylinder='1' o2='33.0%' he='39.0%' />
  <event time='42:12 min' type='20' name='PO2' />
  <event time='42:12 min' type='25' value='80' name='gaschange' />
  <event time='62:08 min' type='25' value='2621455' name='gaschange' />
  <event time='67:10 min' type='25' value='80' name='gaschange' />

```
v3:
```
  <event time='0:02 min' type='25' value='2752527' name='gaschange' cylinder='0' o2='15.0%' he='42.0%' />
  <event time='0:02 min' type='20' name='PO2' />
  <event time='21:56 min' type='20' name='PO2' />
  <event time='21:56 min' type='25' value='2555937' name='gaschange' cylinder='1' o2='33.0%' he='39.0%' />
  <event time='42:12 min' type='20' name='PO2' />
  <event time='42:12 min' type='25' value='80' name='gaschange' cylinder='2' o2='80.0%' />
  <event time='62:08 min' type='25' value='2752527' name='gaschange' cylinder='0' o2='15.0%' he='42.0%' />
  <event time='67:10 min' type='25' value='80' name='gaschange' cylinder='2' o2='80.0%' />

```

Is this a bug?

mturkia, 2017-09-08:
Would it be OK to send the log file privately to few developers? That would be the easiest for us to investigate what is really going on.

There are a few items in your conclusions that can be explained and others just seem to be bugs.

The gas change event has included the o2 and he in the value parameter, but in the newer format it is decoded into human readable values. But of course the value should not change in the conversion to v3 format. On the other hand the gas change event in the beginning of the dive is unnecessary as we know that if you have one cylinder you did breath from it. Besides it was automatically generated and not provided by DC, thus I do not consider it as data loss but clearing up unnecessary information.

There has been development in the sidemount part quite recently and I believe it this has not yet been included in a release version of Subsurface. Thus it would be better to test sidemount issues with the development version and see what still needs to be improved.

antiker1, 2017-09-09:
> Would it be OK to send the log file privately to few developers? That would be the easiest for us to investigate what is really going on.

Sure. where should I send it to?

> But of course the value should not change in the conversion to v3 format. 

Interestingly, the human readable values seemed to have been right (at least IIRC).

> On the other hand the gas change event in the beginning of the dive is unnecessary as we know that if you have one cylinder you did breath from it. Besides it was automatically generated and not provided by DC, thus I do not consider it as data loss but clearing up unnecessary information.

Well point is, as said, one may simply not have any data about the tanks, but still know that one did use tanks (in contrast to e.g. free diving, where gas changes do not occur per se).
So for such cases it was nice to have the gas change event, explicitly indicating that e.g. air was breathed.

The best solution would perhaps the following:
Let people define an cylinder without values (if these are not known, except the type of gas) and one could use such (with the gas change event) for that.
And if no cylinders ?

re defined at all (and thus the gas change event is automatically removed (which I still think should not happen automatically)) then no tanks were used and the dive can only be a free dive (or something else special).

Are there PPA package for such dev versions?

mturkia, 2017-09-09:
Sure. where should I send it to?

miika dot turkia at gmail dot com
dirk at hohndel dot org

    Well point is, as said, one may simply not have any data about the tanks, but still know that one did use tanks (in contrast to e.g. free diving, where gas changes do not occur per se).
    So for such cases it was nice to have the gas change event, explicitly indicating that e.g. air was breathed.

Selecting dive type should pretty much indicate whether there was a tank or not. I just don't recall when this feature was introduced, but it is available on the latest Subsurface.

    Are there PPA package for such dev versions?

https://launchpad.net/~subsurface/+archive/ubuntu/subsurface-daily

antiker1, 2017-09-09:
I've just sent off the mail. Hope that helps.

> Selecting dive type should pretty much indicate whether there was a tank or not. I just don't recall when this feature was introduced, but it is available on the latest Subsurface.

Ah didn't know that was possible in the meantime. Nice.
Still how would I set a dive, for which I know the gas (e.g. air) but have no other data about the tanks?

Any idea about the bogus added time/duration 00:00:00 respectively 00:00? :-)

And what exactly changed in the handling of dives with side-mount tanks with the same gases? I mean how is one intended to set such dives properly?

Can it be that anything is wrong with the repo keys at the daily archive?
I get a "W: Failed to fetch https://launchpad.net/~subsurface/+archive/ubuntu/subsurface-daily/dists/trusty/Release: Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown. The certificate chain uses insecure algorithm.  Could not handshake: Error in the certificate verification."

Thanks!

(sorry, if my answers take a while, I'm on diving vacation :D )

antiker1, 2017-09-09:
btw: I have another case in which setting the gas changes is unclear.

Back then when I did the TMX courses, we made several dives in which I had a number of stage tanks with different gases, and we actually changed between them (many times), not because it would be necessary for the depth or so, but simply for getting experience in the gas change.

For such dives, I have no records on when/which gas I used, only the respective tanks, gases, start and end pressures.

Right now subsurface then stores a single gaschange event, with that of the cylinder 0.

In the end I think it's the same situation as with multiple stage tanks with the same gases:
I want to tell it, that all these were used, but there are no records of which tank was used when.

antiker1, 2017-09-09:
And yet another similar case:
I have dives in which I had stages with different gases, but from e.g. one type of gas, say EAN30 I had two tanks (and both were used).
My dive computer wouldn't even allow me to mark which tank I switched to at which time (only to which gas), however subsurface implicitly adds e.g. `cylinder='1'` which is of course only half the truth.

antiker1, 2017-09-09:
Also related to gas changes/cylinders:
I have dives in which I had tanks which weren't used (in terms of breathing from them). So the start/end pressured is the same.

Subsurface shows these values in red in the UI (why? I think this should be totally valid).
Such case must of course also be considered when doing the multi-tanks with same gas scenario. I.e. one cannot just blindly add all cylinders with the same gas as being used in a gas change event, but only those which which have lower end pressure or which the user/computer either manually marked as used (e.g. in case he hasn't set the pressures at all, but still wants to mark more as used).

mturkia, 2017-09-10:
It is getting difficult to track and respond due to so many problem reported in single issue (with multiple comments)...

The change in value in the gaschange events is related to the cylinder configuration. It appears that the old value is conflicting with the one on cylinder configuration and thus the cylinder config is respected.  I.e. the value was stating that gas was 15/40 when the cylinder was configured for 15/42. And the cylinder configuration is respected when resolving the conflict (as that is what user changes when configuring the gas mixes).

There is some discussion about side-mount support in the mail thread (continues in August):
http://lists.subsurface-divelog.org/pipermail/subsurface/2017-July/thread.html#29404
This might also take care of your TMX course issue, as the logic of cylinder usage was changed quite a bit.

I was just able to install subsurface to Trusty (from the daily PPA) without any certificate errors. Might be that Launchpad had briefly some certificate issues that have been resolved. However, it seems that the compilation does not work anymore and thus the version is somewhat old, Aug 1.

antiker1, 2017-09-13:
Hey.

Sorry for the many problems in one bug, it wasn't obvious to me from the beginning that these are all different.

I see dirkhh has already addressed two problems. And I'll have a look at the mailing list thread you've mentioned when I find some time between diving ;-)

As for the changed value of the gasevent:
What happened here was probably the following:
In the computer, I set the "nominal" mix, e.g. 15/40, which was then probably taken for the gaschange event.
In subsurface (after the dive) I set the actual values as measured with O2/He sensors.
I think it does make some sense that one can store the different two values. The ones for the gaschange event will be that what the computer had used, so it makes sense to store them for later reference on what the computer calculated.
The actual value (as stored in the cylinders) does also make some sense to have it stored, as e.g. subsurface could calculate an even more exact deco profle and show that for comparison with what the computer gave.

The conflict resolution seems however inferior, doesn't it?
These values for o2="" and he="" go into the gaschange event, as such, the values from the gaschange event itself are probably the most canonical ones and should be used, or not?

Any idea already about the messed up depth values?

Thanks!

mturkia, 2017-09-13:
Those were actually my fixes that Dirk merged into master (thus it appears that he did the fixing :D)

I am not certain what would be the correct way to handle the gas mix issue. The value in cylinder configuration comes from the DC initially and if user changes that, he intends the new value to be the correct gas mix. And to me that implies that DC had gas mix set incorrectly, thus we can ignore the DC provided value. Nowadays when cylinder indexing is implemented, the value/o2/he information in gaschange event is not necessary, as we associate the gaschange to cylinder using the index. Thus the value/o2/he information could be removed from the gaschange event altogether.

Subsurface considers the negative depths to be faulty and interpolates a value based on previous and next depth reading instead.

antiker1, 2017-09-14:
Then thanks to you :)

Hmm well not sure about that... IMO it's in principle worth to keep the data from the DC recorded, as this is also what likely determined the actual dive and at least the DC deco stops which subsurface also stores.
It's okay to allow the user to correct the value (as e.g. I did) but this shouldn't override/remove the DC provided data (as it wouldn't just re-compute and override the data for the deco stops using subsuface's deco algorithm).

So first, I think you should keep the values (from the DC) in the event.
I'd probably say: let `o2` and `he` be the values from the DC and drop `value`.
This would still allow you to ignore these DC-provided numbers and take what the user manually set in the cylinders, for whatever subsurface does in terms of further calculations.

Sorry, but interpolating data seems just plain wrong. There's no use in it all all, apart from wiping out possibly reasonable data (e.g. a DC could continue to run for a while after the water has been left and report pressure/depth values relative to that, as the user e.g. drives upon a hill).
It's perfectly fine if you internally ignore such data, or consider it implicitly to be 0m, but anything beyond that not just looses data, but actually makes it up.

Thanks :)

antiker1, 2017-09-20:
Anything new here? This interpolation thingy is pretty annoying. It's a bit like as if graphic software would automatically apply some vivid filters on my images, just because their developer thinks it makes them better ;-)

mturkia, 2017-09-23:
I am not familiar with that part of the code, nor the design decisions behind them. So someone else should be commenting on the remaining issues.

dirkhh, 2017-09-23:
@antiker1 the nice thing with open source software is that no one forces you to use it, and anyone is invited to make it better. The decisions that are reflected in the current code were made based on data that we have seen from users' dive computers - and we made the best decisions we could in order to give the users reasonable data.
Telling us that we are "plain wrong" I'm sure feels good, but is unlikely to motivate anyone to sit down and spend time to try to figure out how to address the issue.

I'm closing this issue as not constructive.

antiker1, 2017-09-23:
@dirkhh same goes probably for the situation when people report issues, thus trying to make the program better, and are then blocked off because of feeling being offended without much reason, doesn't it?

Anyway I didn't want to offend you or any other developer, if you may have felt this was the case by saying it's plain wrong.

It just don't know any other program (be it open source or proprietary) which “auto-corrects” user data to something which it doesn't (and cannot) know whether it's correct, without even asking the user (or by without manual user action). Neither does vim or libreoffice auto-correct spelling if they think it's wrong (unless the user invokes the auto correction), nor does mplayer change the gain of audio files it plays to something it deems better, nor does gimp "correct" the exposure of images without user action.

So what is the benefit of changing data (which is likely simply correct) to something that has never been really measured?

Would it be difficult to change subsurface to not alter the data and simply use 0 (or some interpolation) internally, if it needs to? :-)
Or could one simply ask the user whether he wants such values to be auto-corrected or not?

Thanks

dirkhh, 2017-09-24:
That is a very valid argument. I think having negative values in the samples might break our on disk format - but I'll have to validate that assumption.
I'll also have to go back through the git history to remind myself why we interpolate instead of just using 0.

antiker1, 2017-09-24:
Thanks for looking again into this.

Naively I'd suggest something like:
- store the data as delivered by the DC and don't modify that values (one never knows if some divecomputer may actually do something useful with such values, and thus that data should be dropped)
- if needed for any internal calculations or drawing graphs and that like, take any negative values simply as 0, but I'd be fine with the interpolation thingy as well, I'd just prefer that the "real data" is not lost.

:-)

If you need further logs or help from the user side, don't hesitate to ask!

antiker1, 2017-09-24:
Is there an easy quick way to disable that interpolations/etc. (e.g. disabling something in the code)?
My OSTC4 stores only so and so many dives (not sure how many) internally, and I'd like to get them extracted with values "as is" before the internal logs are overwritten.

mturkia, 2017-09-25:
You might want to save the dumpfile (raw memory of DC) or use libdivecomputer's dctool to store the original info.

glance-, 2017-09-25:
Doesn't work with a OSTC4. There is no memory dump functionality there. You can save the dive binaries with dctool tho.

Anyway. Subsurface manages the data as subsurface see fit. If you edit the data in subsurface, it will edit the saved data to. I don't tink storing multiple versions of the data is the right way to go.

Negative depth are obviously bogus and I see no reason for saving bogus data.

One idea discussed somewhere was to save the dive in prestine format, just as the dc represents it, in the subsurface log book to. That way you can save exactly the bits that the dc gave you, and you can parse the dive again later, to import new information which wasn't imported back in the initial import. As far as I know, no one else was thrilled by this idea, and no one cared enough to implement it. I still think it would be a cool feature.

antiker1, 2017-09-25:
> I don't tink storing multiple versions of the data is the right way to go.

Well, that's why I suggested to store it as it comes from the computer, and if you graphing/calculations don't work with negative numbers take them as zero (without storing them as such).

> Negative depth are obviously bogus and I see no reason for saving bogus data.

Why? It could simply be height above sea level (or rather above the surface level of the last dive).
And even if one knows for a specific model of computer that it's just bogus for that, it's IMO still not ideal to auto-correct it.
Take again the example of photography: One commonly knows that pixels with luminosity of 0 or 255 (in 8 bit channels) are bogus, still no GIMP or rawtheerape automagically, computes&sets something else for these pixels.

> One idea discussed somewhere was to save the dive in prestine format, just as the dc represents it, in the subsurface log book to. That way you can save exactly the bits that the dc gave you, and [...]

Sounds like a great idea, even if it's not directly in the XML, but as separate binary files.

glance-, 2017-09-25:
Negative depth are most likely a calibration issue with the sensor. There is just to much madness going on in the world of dive computers to clean the data up when handling it. 

And yea, everybody know that the best sound comes from wav files.

Subsurface is primarily a tool for handling your logbook, not handling the exact data from your dive computer. If you would like to change that you're free to write the code.

antiker1, 2017-09-26:
> Negative depth are most likely a calibration issue with the sensor. There is just to much madness going on in the world of dive computers to clean the data up when handling it.

Well, everyone is of course entitled to his own opinion :-)

I, personally, don't see much sense in "correcting" such data. One cannot really know for all devices whether it's a calibration issue or not and thus destroys possibly valid data.
To the least, if it's a calibration issue, then most likely the other depth values close to 0.0 are incorrect as well, and thus interpolating new values for the negative ones from these doesn't seem that useful to me.

Also, but I haven't looked into the code, I don't see the big difference in cleaning up data on writing to cleaning up data on reading for things like graphs and calculations.

> Subsurface is primarily a tool for handling your logbook, not handling the exact data from your dive computer. If you would like to change that you're free to write the code.

Is there any good reason not the handle exact data in the logbook?
And I have no intention on forking subsurface ;-) and since you already implied you're against keeping the data as is, it would boil down to that, wouldn't it? I'd rather simply use another software, even if that was regrettable. :-(

I've just reported the whole thing in good faith, since I think it's bad to change/"correct" such data without user consent and assumed it was simply a bug and not intended behaviour.
Didn't expect to such strong desire to keep that.

Best wishes :-)

mikeller, 2026-10-17:
Hi @antiker1,

Revisiting this long thread: the v2-to-v3 migration behaviours you raised are deliberate design choices, not corruption, redundant single-cylinder gaschange events with no gas data carry no information and are dropped; negative depth values are near-universally sensor calibration artefacts and are smoothed out rather than kept as-recorded, since there's enough inconsistency across dive computers' raw data that cleaning it up case-by-case isn't practical. The idea of keeping each dive's pristine as-received DC data alongside Subsurface's interpreted version came up as an interesting alternative, but never gained enough interest from anyone to be implemented.

Closing this per that design explanation. Please reopen with a specific, current-build case if you believe one of these transformations is genuinely destroying information you need, rather than smoothing/dropping redundant data as intended.

closed 2026-10-17 by mikeller