subsurface/subsurface #562: Tankbar info wrong, no pressure info for 2nd, 3rd,... gas, vertical lines in pressure graphs

sfuchs79, 2017-09-09:
Found a few issues which were introduced between ~4.6.4.390 and now (~4.6.4.790):
- The tankbar info in the graph is incorrect for all of my logged or planned dives. Tankbar always shows 1st gas.
- In the infobox of the graph a pressure value is displayed only for the 1st gas while at the same moment there is a correct pressure graph also for the 2nd, 3rd,... gas.
- Sometimes there are some incorrect vertical lines connecting the pressure graph of two gases.

[image]

[image]

sfuchs79, 2017-09-11:
Also the tissue heatmap is not correct any more. Also here no other gas then 1st gas is considered.

dirkhh, 2017-09-13:
My guess is that these are all side effects from some of Linus' changes to the tank handling.
Thanks for reporting them. Now someone needs to track them down and fix them :-/

dirkhh, 2017-09-14:
The tank bar and info box are now fixed.
Linus asked for a sample dive that shows the artifact in the picture
and @atdotde might want to look at the heat map :-)

dirkhh, 2017-09-23:
Ha, @neolit123 - no, this one is now assigned to @atdotde 
The heatmap is his baby :-)

atdotde, 2017-10-17:
@sfuchs79 can you give me a concrete example of a dive where the heat map does not follow gas changes with current master? I cannot reproduce this bug anymore.

sfuchs79, 2017-10-17:
Current master:
[image]

4.6.4:
[image]

Looks definitively different - Maybe the question is: What is correct and what is wrong? Maybe even 4.6.4 was wrong?!

sfuchs79, 2017-10-17:
Hmmm.. honestly I have to admit that current master heatmap looks "nicer" compared to 4.6.4. But have we been aware of the fact that it was broken in 4.6.4 and you then fixed it?
If this is the case then: Sorry, I was too focused on the difference and was immediately deciding: 4.6.4 good (what else?) - current master broken after the changes Linus did...
@atdotde But I'm waiting for your final conclusion now...

glance-, 2017-10-17:
To me it looks like master renders the head map with the back gas all the way, and that's not right.

PR #680 (2017-10-18): Find current gasmix for heatmap
To compute the heatmap value, we need the current gasmix but
the current cylinderindex is no longer available.

I still think this approach is worse than #679 since it walks the events list for every pixel of the heat map but fine. Trade performance for memory footprint.

Fixes #562

Suggested-by: Linus Torvalds <torvalds@linux-foundation.org>

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

## [... cut]
commit 432110ac (2017-10-18): Find current gasmix for heatmap

To compute the heatmap value, we need the current gasmix but
the current cylinderindex is no longer available.

Fixes #562

Suggested-by: Linus Torvalds <torvalds@linux-foundation.org>
closed 2017-10-18 by dirkhh